SensorsINI/v2e #1: Events duration does not match input video duration

enrico-c, 2020-07-02:
Hi,

When exporting the events in HDF5 format, the timestamp of the last event does not correspond to the end time of the input video, but it is scaled by the input_slowmotion_factor. However, the event stream and input video should have the same duration.

Here is an example.
From the log: video has total 1167 frames with total duration 23.32s. 
Here I use a slomo factor of 3. The last event timestamp in events.h5 is 7780000, in us.
Multiply back the time of events by the slomo factor to obtain the total duration of the input video: 7780000/1e6*3 = 23.3 seconds 

To get the right event timestamps is it enough to scale them up by the slomo factor?

Thanks

duguyue100, 2020-07-02:
Ok, now I realized the confusion here. You must set `--input_slowdown_factor 3`, this argument is designed to determine the actual frame rate by the user. Say an input video is sampled at 120fps yet presented as a 30fps video. The `--input_slowdown_factor` should be 4.

The actual slowdown factor is determined by the `timestamp_resolution`. So in your case, when you set `input_slomotion_factor` to 3, it means the actual frame rate the `v2e` understand is `1167/23.32*3` which is about 150fps.

You can leave the `input_slomotion_factor` as default, and then the time stamp should be fine.

tobidelbruck, 2020-07-02:
Thanks Yuhuang, I updated the option help to make it more clear.

enrico-c, 2020-07-02:
I see, I was assuming the `input_slomotion_factor` was related to a previous parameter to set the number of frames to be interpolated (the actual slow motion factor), but that is not the case.
Thanks for clarifying, I'm closing the issue.

closed 2020-07-02 by enrico-c