SensorsINI/v2e #2: Timestamps of the generated event stream

RunqiuBao, 2020-07-15:
hi, thanks for this great simulator! It is really convenient to use and noise models look cool.

One problem is that, it seems all the events between two interpolated frames have same timestamps. 
For example, when I set --timestamp_resolution=0.0005 (1920fps). Then within 0.0005s, all the events have same timestamps.
Is it? Or am I not using it correctly?

I suppose to generate events, it first interpolates the video to 1920fps. Then, events are generated by comparing intensity difference between two frames.  
My point is that these events between two frames can still have different timestamps by linear interpolation, which will make the simulator more realistic.

tobidelbruck, 2020-07-16:
Thanks for the positive feedback.

About the timestamps, yes, that is how they are currently generated, to be aligned with the generated frames except when there are multiple events generated by the new frame. Then the events are spread between the frames. The result is a bit weird looking pillars of events as illustrated in the animated GIF screenshot of 3D spacetime display below (not sure if this makes it to github comments).

We could randomly scatter all events between frame times. This would reduced synchronicity but it would be fake too, since the choice of exact timestamp would be arbitrary. Or do you have some other suggestion?  Anyhow, we can add the option of randomizing the timestamp. It might help for developing noise filters that rely on coherence to filter in signal events.
[image]

RunqiuBao, 2020-07-17:
Hi, I saw the source code about generating timestamps. And I have one suggestion.
So I noticed that there is a class member variable called self.baseLogFrame in class EventEmulator which stores the per-pixel log intensity since last triggering at every pixel. 
However, only a uniform timestamp is stored for self.baseLogFrame called t_previous.

I suppose if there is another member variable to store per-pixel t_previous, then the actual triggering time of a recent event at a certain pixel can be estimated by linear interpolation.
Do you think if it is a viable idea?

tobidelbruck, 2020-07-19:
Really sorry about delayed response; got busy coding stuff for our [l2race challenge](https://sites.google.com/view/telluride2020/challenges?authuser=0) for Telluride neuromorphic workshop.

Your approach might work. v2e currently upsamples (or not depending on option) from source frames, and it is only at these frame times that it estimates the input intensity and computes whether there are events there. So events are generated at these moments, unless there are more than 1 event, in which case we use the strategy as described in our arxiv paper. 

The simplest approach to decorrelate these events is to assign them a random time between the frames. It is complicated by the need to output the events monotonically in time order, so generated events would need to be sorted into monotonic time order.

What do you mean by interpolation?  You mean linearly or quadratically or spline interpolating the intensity samples to be able to further predict the times of intermediate events? That might work, but I don't see how to do it right now. It would be great if you could implement this.
Tobi

RunqiuBao, 2020-07-20:
Thanks for the prompt response, Prof. Tobi.
I understand your concern and will try to implement and see.

duguyue100, 2020-08-19:
I will close this issue for now.

closed 2020-08-19 by duguyue100