SensorsINI/v2e #3: Error running code

xlj199, 2020-07-27:
Hi, I have encountered an error running the code.
I am running on win10 python3.7.An error is as follows:
............................
WARNING:__main__:Gooey GUI builder not available, will use command line arguments.
Install with "pip install Gooey". See README
WARNING:__main__:Gooey GUI not available, using command line arguments.
You can try to install with "pip install Gooey"
ERROR:__main__:specify either --overwrite or --unique_output_folder
..............................

tobidelbruck, 2020-07-27:
It's possible there is a bug with the options since it was hard to test 
all combos.

What was your command line exactly? Did you supply or the other of

ERROR:*main*:specify either --overwrite or --unique_output_folde

Thanks

xlj199, 2020-07-28:
> It's possible there is a bug with the options since it was hard to test all combos. What was your command line exactly? Did you supply or the other of ERROR:*main*:specify either --overwrite or --unique_output_folde Thanks

Thank you for your reply.
I used what you provided sample input data (included SuperSloMo model and tennis.mov).
I'm going to put these in a folder called Input.
Then, Use the command in cmd  "python v2e.py -i input/tennis.mov --timestamp_resolution=.005 --dvs_exposure duration 0.005 --output_folder=output/tennis --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2 tennis.aedat --output_width=346 --output_height=260 --stop_time=3"
I got the error.

duguyue100, 2020-07-28:
Your command looks fine to me. I see you are using Windows to run v2e, could it a path problem? Maybe you can try to substitute input and output folders with absolute paths and try again?

xlj199, 2020-07-28:
> Your command looks fine to me. I see you are using Windows to run v2e, could it a path problem? Maybe you can try to substitute input and output folders with absolute paths and try again?

I solved the problem! 
Remove "--overwrite" from the command line.Thanks for the tip.
One more question，I found that the generated video is not the original video before the conversion(In terms of time and speed of the video).If I want to get the same length and frame rate of video, how do I set the parameters?

tobidelbruck, 2020-07-28:
Which generated video? Do you mean the DVS video? If you refer to DVS 
video, yes, it will not be same length as input video in general since 
it might be exposed by constant_duration, constant_count or area_count 
methods.

And frame rate of constant_duration DVS frame output video is (I recall) 
taken from frame rate of input video, but the DVS video is slowed down 
by the frame interpolation.

Could you provide a bit more specific example? And look carefully at the 
extended usage output from --help invocation.

Thanks

xlj199, 2020-07-28:
> Which generated video? Do you mean the DVS video? If you refer to DVS video, yes, it will not be same length as input video in general since it might be exposed by constant_duration, constant_count [...]

Yes，it is dvs-video.avi.
I used the example video you mentioned(tennis.mov ),it has 25 seconds.However, The generated DVS video is only 19 seconds long and the frame rate is noticeably slower.
Well,let me take a closer look at the help.

tobidelbruck, 2020-07-28:
It could well be a bug, thanks for noticing.

I'm extremely busy now with coding up something for the Telluride 
workshop and can't spend time right now to look. I can look after next 
week. Let me know if you need some hints.

xlj199, 2020-08-06:
> It could well be a bug, thanks for noticing. I'm extremely busy now with coding up something for the Telluride workshop and can't spend time right now to look. I can look after next week. Let me know if you need some hints.

Hi, thank you for your clarification.
Now I have another problem.I have some data in .aedat2.0 format and want to convert it to .txt format.
Is there a script for such a simple format conversion?(Actually, I tried and failed.My answer seems to be (x,y), no t)

duguyue100, 2020-08-06:
@xlj199 there is no easy fix as if you want to load and decode AEDAT2.0 then it tends to load the entire file into the memory.

A possible way to do this is through the DV python bindings, it implements AEDAT2.0 API. See here: https://gitlab.com/inivation/dv/dv-python/blob/master/README.md

But you need to write the code to drop all the data into a text file.

xlj199, 2020-08-06:
@duguyue100 I implemented the idea, but the parameters T  seemed hard to understand(It's not starting from zero, and it has no units).

duguyue100, 2020-08-06:
The unit for time is microsecond and in AEDAT2, it's coded into a INT64 number.
As the T not starting from 0, I think it's normal because the events are uniformly distributed between 0 and 2nd frame's time.
At least it should be a number that is close to 0.
Make sure you decode the AEDAT format correctly as in the reference code above.

tobidelbruck, 2020-08-06:
Yes, you can at least output txt from AEDAT2.0 from jaer using 
DavisTextOutputWriter filter. Please see the jaer user guide doc. And if 
not clear, please let me know.

duguyue100, 2020-08-19:
I hope you find the solution, close the issue for now.

closed 2020-08-19 by duguyue100