SensorsINI/v2e #5: some formulation errors

fourson, 2020-08-19:
In emlator.py, line481:
shotOffProbThisSample = shotNoiseFactor*np.divide(
                        self.pos_thres_nominal, self.pos_thres)
may be :
shotOffProbThisSample = shotNoiseFactor*np.divide(
                        self.neg_thres_nominal, self.neg_thres)

duguyue100, 2020-08-19:
You are right, let me fix it right away, thanks!

fourson, 2020-08-19:
and line516:
self.baseLogFrame[shotOffCord] += \
                            shotOffCord[shotOffCord] * \
                            self.neg_thres[shotOffCord]
may be 
self.baseLogFrame[shotOffCord] -= \
                            shotOffCord[shotOffCord] * \
                            self.neg_thres[shotOffCord]

fourson, 2020-08-19:
line 409 410
pos_cord = (pos_frame > self.pos_thres * (i + 1))
neg_cord = (neg_frame > self.neg_thres * (i + 1))
">" should be ">="

tobidelbruck, 2020-08-19:
Thanks, we will fix it. Yuhunag, since you just pushed another fix, 
would you mind checking this one?

On 8/19/2020 2:08 PM, silhouette wrote:
>
> line 409 410
> pos_cord = (pos_frame > self.pos_thres * (i + 1))
> neg_cord = (neg_frame > self.neg_thres * (i + 1))
> ">" should be ">="
>

duguyue100, 2020-08-19:
Thanks, line 409, 410, 516 has been corrected. thanks!

tobidelbruck, 2020-08-28:
Are you sure about this? The neg_thres is negative, isn't it? The whole 
simulation would not work if we had gotten this wrong.

Tobi

fourson, 2020-08-28:
> Are you sure about this? The neg_thres is negative, isn't it? The whole simulation would not work if we had gotten this wrong. Tobi
> […](#)

Please kindly note that the var "neg_thres" is indeed positive (initialized with " neg_thres=0.2" in line 72, and set to an array in line 212). PS: in your comment (line 541), it was write as :"# neg_thres is >0"

tobidelbruck, 2020-12-19:
Seems to be OK

closed 2020-12-19 by tobidelbruck