SensorsINI/v2e #6: error launching DDD example

jiemdev, 2020-10-06:
Hello,

I am trying to run DDD example and I get following error:

(v2e) $ python -m dataset_scripts.ddd.ddd-v2e.py --input DavisDrivingDataset/rec1501902136.hdf5 --slomo_model input/SuperSloMo39.ckpt --slowdown_factor 20 --start 70 --stop 73 --output_folder v2e-output/ddd20-v2e-short --dvs_aedat dvs --pos_thres=.2 --neg_thres=.2 --overwrite --dvs_vid_full_scale=2 --frame_rate=100
Traceback (most recent call last):
  File "v2e/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "v2e/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "v2e/dataset_scripts/ddd/ddd-v2e.py", line 27, in <module>
    from v2e.v2e_utils import OUTPUT_VIDEO_FPS, all_images, \
ImportError: cannot import name 'OUTPUT_VIDEO_FPS' from 'v2e.v2e_utils' (v2e/v2e/v2e_utils.py)

Then I try and hardcode OUTPUT_VIDEO_FPS = 30 in the code, and I get new error:
(v2e) $ python -m dataset_scripts.ddd.ddd-v2e.py --input DavisDrivingDataset/rec1501902136.hdf5 --slomo_model input/SuperSloMo39.ckpt --slowdown_factor 20 --start 70 --stop 73 --output_folder v2e-output/ddd20-v2e-short --dvs_aedat dvs --pos_thres=.2 --neg_thres=.2 --overwrite --dvs_vid_full_scale=2 --frame_rate=100
usage: -m [-h] [-o OUTPUT_FOLDER] [--overwrite]
          [--unique_output_folder UNIQUE_OUTPUT_FOLDER] [--no_preview]
          [--avi_frame_rate AVI_FRAME_RATE]
          [--auto_timestamp_resolution AUTO_TIMESTAMP_RESOLUTION]
          [--timestamp_resolution TIMESTAMP_RESOLUTION]
          [--output_height OUTPUT_HEIGHT] [--output_width OUTPUT_WIDTH]
          [--dvs_params DVS_PARAMS] [--pos_thres POS_THRES]
          [--neg_thres NEG_THRES] [--sigma_thres SIGMA_THRES]
          [--cutoff_hz CUTOFF_HZ] [--leak_rate_hz LEAK_RATE_HZ]
          [--shot_noise_rate_hz SHOT_NOISE_RATE_HZ]
          [--dvs128 | --dvs240 | --dvs346 | --dvs640 | --dvs1024]
          [--disable_slomo] [--slomo_model SLOMO_MODEL]
          [--batch_size BATCH_SIZE] [--vid_orig VID_ORIG]
          [--vid_slomo VID_SLOMO] [--slomo_stats_plot] [-i INPUT]
          [--input_slowmotion_factor INPUT_SLOWMOTION_FACTOR]
          [--start_time START_TIME] [--stop_time STOP_TIME]
          [--dvs_exposure DVS_EXPOSURE [DVS_EXPOSURE ...]] [--dvs_vid DVS_VID]
          [--dvs_vid_full_scale DVS_VID_FULL_SCALE] [--dvs_h5 DVS_H5]
          [--dvs_aedat2 DVS_AEDAT2] [--dvs_text DVS_TEXT]
          [--dvs_numpy DVS_NUMPY] [--rotate180 ROTATE180] [--numpy_output]
-m: error: unrecognized arguments: --slowdown_factor 20 --frame_rate=100

Then I give a try without thses arguments and here the result:
(v2e) $ python -m dataset_scripts.ddd.ddd-v2e.py --input DavisDrivingDataset/rec1501902136.hdf5 --slomo_model input/SuperSloMo39.ckpt --start 70 --stop 73 --output_folder v2e-output/ddd20-v2e-short --dvs_aedat dvs --pos_thres=.2 --neg_thres=.2 --overwrite --dvs_vid_full_scale=2
v2e/bin/python: Error while finding module specification for 'dataset_scripts.ddd.ddd-v2e.py' (ModuleNotFoundError: __path__ attribute not found on 'dataset_scripts.ddd.ddd-v2e' while trying to find 'dataset_scripts.ddd.ddd-v2e.py')

Do you plan to update these scripts?

Regards

tobidelbruck, 2020-10-06:
Thanks for the bug report.  I confess that we haven't looked at the DDD 
part of v2e for some time.  We'll try to have a look and fix it so the 
examples work again.  The basic problem is that handling the DDD20 data 
is very different than reading frames from video and so there is a 
completely separate script for dealing with the DDD data. But it is 
likely that changes to underlying utilities or shared args broke DDD20. 
Sorry about that. Will have a look ASAP.

Tobi

tobidelbruck, 2022-05-18:
I think we will abandon support for the ddd-v2e.py script. However, I just pushed a fix to the ddd_extract_data.py script and added it to setup.py. I tested on one DDD20 recording and it successfully extracted DVS events and APS frame video file. Updated README.md.

closed 2022-05-18 by tobidelbruck