SensorsINI/v2e #7: yml fails to reproduce

satabios, 2020-10-09:
pip fails to build using the provided yml file

duguyue100, 2020-10-09:
@satabios Hi, can you elaborate this error? some error snapshot would be very helpful.

tobidelbruck, 2020-10-09:
The yml file is for using conda environment construction, not for pip.  
See 
https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html

Since conda environment files are not yet supported by pycharm, what I 
do is to make a conda env and then use pip from command line (or using 
pycharm's support of requirements.txt files via requirments plugin) to 
install the libraries.

There is no ideal solution yet to this greatest weakness of python.

Tobi

satabios, 2020-10-09:
> @satabios Hi, can you elaborate this error? some error snapshot would be very helpful.

I tried to install the environment on conda using `conda env create -f environment.yml`:

`Collecting package metadata (repodata.json): ...working... done
Solving environment: ...working... done
Preparing transaction: ...working... done
Verifying transaction: ...working... done
Executing transaction: ...working... done
Ran pip subprocess with arguments:
['/media/14tb/sathya/anaconda/envs/v2e/bin/python', '-m', 'pip', 'install', '-U', '-r', '/media/14tb/sathya/event-based/v2e/condaenv.q8s_8h_4.requirements.txt']
Pip subprocess output:
Collecting engineering-notation==0.6.0
  Using cached engineering_notation-0.6.0-py3-none-any.whl (5.7 kB)
Collecting Gooey
  Using cached Gooey-1.0.4.0.0.0-py2.py3-none-any.whl (617 kB)
Processing /home/lubos/.cache/pip/wheels/2d/43/97/00701864a7bee6d9e1a52dd682537dcbf1d013d0e2e6f0c1f1/psutil-5.7.2-cp37-cp37m-linux_x86_64.whl
Collecting wxpython==4.0.7
  Using cached wxPython-4.0.7.tar.gz (68.9 MB)
Requirement already satisfied, skipping upgrade: Pillow>=4.3.0 in /media/14tb/sathya/anaconda/envs/v2e/lib/python3.7/site-packages (from Gooey->-r /media/14tb/sathya/event-based/v2e/condaenv.q8s_8h_4.requirements.txt (line 2)) (7.2.0)
Processing /home/lubos/.cache/pip/wheels/5f/e1/fb/d0e85a8383ff58962319bb81c46e398fa1f4bb9e1feb0f81c4/colored-1.4.2-py3-none-any.whl
Requirement already satisfied, skipping upgrade: six in /media/14tb/sathya/anaconda/envs/v2e/lib/python3.7/site-packages (from wxpython==4.0.7->Gooey->-r /media/14tb/sathya/event-based/v2e/condaenv.q8s_8h_4.requirements.txt (line 2)) (1.15.0)
Requirement already satisfied, skipping upgrade: numpy in /media/14tb/sathya/anaconda/envs/v2e/lib/python3.7/site-packages (from wxpython==4.0.7->Gooey->-r /media/14tb/sathya/event-based/v2e/condaenv.q8s_8h_4.requirements.txt (line 2)) (1.19.1)
Building wheels for collected packages: wxpython
  Building wheel for wxpython (setup.py): started
  Building wheel for wxpython (setup.py): still running...
  Building wheel for wxpython (setup.py): finished with status 'error'
  Running setup.py clean for wxpython
Failed to build wxpython
Installing collected packages: engineering-notation, psutil, wxpython, colored, Gooey
    Running setup.py install for wxpython: started
    Running setup.py install for wxpython: finished with status 'error'`

tobidelbruck, 2020-10-09:
Yeah, gooey is a problem for Linux. You don't need it. Just comment out.
-- 
Tobi Delbruck, www.ini.uzh.ch/~ tobi, Swiss cell +41766291500, USA cell +16265102646

satabios, 2020-10-13:
Even after commenting it, pip fails to build

`Running command: build_py
    Checking for /tmp/pip-install-50jlvutn/wxpython/bin/waf-2.0.8...
    "/media/14tb/sathya/event-based/v2e/v2e/bin/python" /tmp/pip-install-50jlvutn/wxpython/bin/waf-2.0.8 --wx_config=/tmp/pip-install-50jlvutn/wxpython/build/wxbld/gtk3/wx-config --gtk3 --python="/media/14tb/sathya/event-based/v2e/v2e/bin/python" --out=build/waf/3.7/gtk3 configure build
    Setting top to                           : /tmp/pip-install-50jlvutn/wxpython
    Setting out to                           : /tmp/pip-install-50jlvutn/wxpython/build/waf/3.7/gtk3
    Checking for 'gcc' (C compiler)          : /usr/bin/gcc
    Checking for 'g++' (C++ compiler)        : /usr/bin/g++
    Checking for program 'python'            : /media/14tb/sathya/event-based/v2e/v2e/bin/python
    Checking for python version >= 2.7.0     : 3.7.8
    python-config                            : /media/14tb/sathya/event-based/v2e/v2e/bin/python3.7-config
    Asking python-config for pyext '--cflags --libs --ldflags' flags : yes
    Testing pyext configuration                                      : Could not build python extensions
    The configuration failed
    (complete log in /tmp/pip-install-50jlvutn/wxpython/build/waf/3.7/gtk3/config.log)
    Command '"/media/14tb/sathya/event-based/v2e/v2e/bin/python" /tmp/pip-install-50jlvutn/wxpython/bin/waf-2.0.8 --wx_config=/tmp/pip-install-50jlvutn/wxpython/build/wxbld/gtk3/wx-config --gtk3 --python="/media/14tb/sathya/event-based/v2e/v2e/bin/python" --out=build/waf/3.7/gtk3 configure build ' failed with exit code 1.
    Finished command: build_py (0m1.11s)
    Finished command: build (0m2.222s)
    Command '"/media/14tb/sathya/event-based/v2e/v2e/bin/python" -u build.py build' failed with exit code 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /media/14tb/sathya/event-based/v2e/v2e/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-50jlvutn/wxpython/setup.py'"'"'; __file__='"'"'/tmp/pip-install-50jlvutn/wxpython/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-koo2l040/install-record.txt --single-version-externally-managed --compile --install-headers /media/14tb/sathya/event-based/v2e/v2e/include/python3.7m/wxpython Check the logs for full command output.

CondaEnvException: Pip failed`

duguyue100, 2020-10-13:
`wxPython` is a GUI library that you don't need essentially. as long as you installed other dependencies, you can run `v2e` from command line, you can ignore this.

tobidelbruck, 2020-10-13:
Yes, that's correct. I will update README to try to make this more clear.

closed 2020-10-15 by satabios