SensorsINI/v2e #8: Hi, which algorithm should I use to parse the transformed File **.aedat?

wangxiao5791509, 2020-12-16:
Hi, thanks for your wonderful work first. My problem is which algorithm should I use to parse the transformed File **.aedat?  I find the obtained aedat file can't be parsed via: 

from dv import AedatFile

with AedatFile(input_filename) as f: 
    events = np.hstack([event for event in f['events'].numpy()]) 

But aforementioned code can be used for videos captured from DVS346 sensors. so, would you please tell me how to load the transformed aedat files? Thank you very much! 

Best regards, 
Xiao Wang

tobidelbruck, 2020-12-16:
Dear Xiao, thanks, we will clarify that the AEDAT files are AEDAT-2.0 files that are compatible with jAER (https://jaerproject.net).

If you activate the option, then V2E will also produce text output file with the DVS events. You should be able to parse this file in python, for example.

```
  --dvs_text DVS_TEXT   Output DVS events as text file with one event per line
                        [timestamp (float s), x, y, polarity (0,1)]. (default:
                        v2e-dvs-events.txt)
```
At the moment I'm using V2E for a project and have added more options, for example to write the output file to the input source video folder. Sorry about the mole of options. I know it looks complicated but this is all stuff we use with V2E.

I plan to add an option to specify a python method input for the frames that will allow generating synthetic input more easily than via intermediate AVI.

wangxiao5791509, 2020-12-16:
Cool, thanks for your kind explanation! I will try this.

tobidelbruck, 2020-12-19:
Some future project is to output AEDAT-4.0

closed 2020-12-19 by tobidelbruck