SensorsINI/v2e #9: How to suppress a specific output file?

ygrayson, 2021-01-06:
I would like to suppress the .aedat file from being generated in the output folder, but it seems to always default to v2e-dvs-events.aedat.
Is there any command line argument that I can use to suppress this output file from being generated?

Thanks

duguyue100, 2021-01-06:
Hi, if you read these lines: https://github.com/SensorsINI/v2e/blob/master/v2e/v2e_args.py#L219-L222

Then you will see that by default, it always output AEDAT files. I will write a custom type function so that when you pass None, then it's suppressed.

Yuhuang.

duguyue100, 2021-01-06:
Ok, please pull the changes, and now pass the following arguments in your command:

```
----dvs_aedat2 None --dvs_text None
```

This will suppress all output files.

ygrayson, 2021-01-06:
Great! Thank you so much!

closed 2021-01-06 by ygrayson