SensorsINI/v2e #10: DVS .npy file not generated as expected

ygrayson, 2021-01-07:
Hello,

I am trying to convert a frame-based video and generate the DVS in numpy .npy format in the output folder. The command line option I am using is the following:

`python v2e.py --slomo_model=input/SuperSloMo39.ckpt --input input/tennis.mov --output_folder=output/tennis --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.05 --cutoff=30 --leak_rate=.1 --shot=.1 --dvs_exposure duration 0.005 --dvs_aedat2=None --dvs_numpy=dvs-numpy.npy --output_width=346 --output_height=260 --batch=32 --stop_time=1`

However, in the output folder, I do not see any dvs-numpy.npy file as expected. And there is no warning about .npy file or anything like this. Any help would be much appreciated.

duguyue100, 2021-01-07:
Yes, if you look at this code block: https://github.com/SensorsINI/v2e/blob/master/v2e/emulator.py#L568-L585
you will find that there is no `numpy` saving option implemented.

I suggest that you switch the saving method to HDF5.
I will probably put out a warning.
Thanks for reporting this issue.

ygrayson, 2021-01-07:
I see. Thank you

closed 2021-01-07 by ygrayson