SensorsINI/v2e #11: Failed to reproduce tennis.mov

ygrayson, 2021-01-07:
Hi,

As I am trying to reproduce the tennis.mov example given in the README. I realized the reproduced result is far from the one shown on the website and lacks stability in the dvs video. I first used the command line options in README:

`python v2e.py -i input/tennis.mov --timestamp_resolution=.005 --dvs_exposure duration 0.005 --output_folder=output/tennis --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2 tennis.aedat --output_width=346 --output_height=260 --stop_time=3
`

Then it prompted error:
`ERROR: __main__: specify either --overwrite or --unique_output_folder`

So I deleted --overwrite then came across this error:
`ERROR: __main__:auto_timestamp_resolution=True and timestamp_resolution=0.005: Disable auto_timestamp_resolution`

Then I set auto_timestamp_resolution to be False. Overall using the following command line options:
`python v2e.py -i input/tennis.mov --timestamp_resolution=.005 --auto_timestamp_resolution=False --dvs_exposure duration 0.005 --output_folder=output/tennis --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2 tennis.aedat --output_width=346 --output_height=260 --stop_time=3
`

The code was able to successfully finish. But the generated DVS video is the following:

https://user-images.githubusercontent.com/33047668/103879127-632bad00-5112-11eb-9890-adb59c14d42a.mov

Is there anything wrong with my command line options or is the current code different from previous version in some way?

Thank you in advance.

tobidelbruck, 2021-01-07:
Thank you very much for sharing this result with detailed command line 
and video link, that saves a lot of time.  I think it is a symptom of 
something I also just ran into, which is caused by using infinite 
photoreceptor cutoff frequency. Can you confirm you still get those 
sustained events like the afterimage even if you set photoreceptor 
cutoff frequency in Hz to finite value, e.g. 100 ?

Yuhuang, I will try to track it down by tomorrow, leave it to me. It's 
well possible that I introduced a "regression" i.e. bug by recent 
improvements to allow generating synthetic input from python module.

Tobi

ygrayson, 2021-01-08:
I included `--cutoff_hz=100` in the command line argument and here is the result I got:

https://user-images.githubusercontent.com/33047668/103967180-6bc8c580-519c-11eb-9265-56bd2928612c.mp4

It seems like the sustained events are still there.

tobidelbruck, 2021-01-08:
I agree, there is definitely a regression (i.e. bug I reintroduced with 
recent improvements). I can also reproduce it even with a very low 
cutoff frequency.  Am trying to track it down.

tobidelbruck, 2021-01-09:
Still debugging it, tricky... deep in pipeline that emulates the DVS 
events. Hang on.

tobidelbruck, 2021-01-09:
There is definitely a bug I put in while fixing a different bug. For 
now, until I can figure this out, you can try to roll back to Oct 13 version

853dc121a4c868dcc98dc560874b9f3f8dd6a04e

https://github.com/SensorsINI/v2e/commit/853dc121a4c868dcc98dc560874b9f3f8dd6a04e

Please let us know if you still get the problem there. I will update 
once I figure it out.

tobidelbruck, 2021-01-09:
Ok, bug in emulator.py fixed I hope, see the log. Made some other 
improvements, I hope. Please let us know.

ygrayson, 2021-01-11:
Hi Tobi,

Thank you so much for fixing this bug. As of now, I pulled your newest changes and the tennis.mov example is able to generate DVS data successfully. I will let you know if I encounter any new problem.

Thanks,
Grayson

closed 2021-01-12 by ygrayson