SensorsINI/v2e #12: Event time stamps do not match with video

DeeperMind, 2021-01-20:
Hello, this is a really awesome tool! Thank you for your wonderful research and development!

After diving deep into the generated events, however, I found that the time stamps of the events do not match with the video timeline. Consider a input video with framerate 30 and Super SloMo is turned on. According to my experiment, the events between 0-33ms correspond to the motion between the first and second frame, which is 33-67ms in the video. On the contrary, the events of the last 33ms correspond to the motion between the second last and last frame of the video. In another word, the events are 33ms ahead of the video at the beginning, but this gap is gradually fulilled, and disappeared at the end.  As a result, I cannot cut the events accurately to match the movement of each frame.

DeeperMind, 2021-01-20:
To be more specific, in my case, the video has 936 frames with framerate 30, totalling 31200ms. The events of 0-33ms corrpesonds to 33-67ms in the video, but events of 31167-31200ms corresponds to 31167-31200ms in the video. which makes me very confused.

tobidelbruck, 2021-01-20:
Thanks Jinan for reporting this!  We'll see if we can reproduce it and 
fix the problem. The handling of the timestamps is a bit messy now, 
needs fixing in any case.  -Tobi

DeeperMind, 2021-01-20:
I think I found that solution. The events are great, only the timestamps are not scaled properly. The timestampe should be rescaled to [1/framerate, endtime] to skip the first frame of the video. The reason for this is that Super SloMo cannot recover the motion of the first frame, thus no events are generated.

tobidelbruck, 2021-01-20:
I'm not sure about the argument below, but I think you are off by one 
frame.

If your video has N frames, then v2e will generate events from frame 0 
up to frame N-1, i.e. N-1 intervals. I.e. if your video had only 2 
frames, v2e would generate DVS events for 1 frame, not 2.  If the 
upsampling is manually set to e.g. 4, then for a 2 frame video, 
upsampling would generate 3 interframes between frame 0 and 1. Please 
see the comments at start of slomo.py, line 243 and following in 
https://github.com/SensorsINI/v2e/blob/846ed68bd283ec6feba09069278c77a3f27041ba/v2e/slomo.py#L243-L267

Could you check the reasoning? And could you please explain a bit more 
the expected vs observed behavior? I want to fix the bug but I don't 
understand it yet.

Thanks

tobidelbruck, 2021-01-20:
Dear Jinan, Cool, I will figure it out and push the fix, thank you for 
figuring out the root of the reason. So the comments in slomo.py are OK? 
It's just the creation of interpTimes in v2e.py that is wrong?

-Tobi

DeeperMind, 2021-01-20:
I do not konw much about how slowdown factor works in the code as I am not using it. As far as  I am concerned, simply adjusting the scale of the timestamps is good enough.

closed 2022-05-18 by tobidelbruck