SensorsINI/v2e #16: File AEDAT can't be converted with Dynamic Vision Viewer

Kyrbouille, 2021-04-22:
Hello,

I have recently been working in an internship which consists of creating a database with DVS images. So I inquired about this tool by having tested it with indications of this github. However, I encountered a problem with the "aedat" file.

Knowing that this version of aedat is no longer up to date, there is a tool in Dynamic Vision Viewer to convert the "aedat" file into "aedat4" file to read it after but there is an error during the conversion. Do you have any idea why this wouldn't work?

Here is the command used about the build: 
`python v2e.py -i input / vid.avi --overwrite --timestamp_resolution = .003 --auto_timestamp_resolution = False --dvs_exposure duration 0.005 --output_folder = output --overwrite - -pos_thres = .15 --neg_thres = .15 --sigma_thres = 0.03 --dvs_aedat2 vid.aedat --output_width = 240 --output_height = 180 --stop_time = 3 --cutoff_hz = 15 --ignore-gooey`

Thank you very much for your time.
Cordially.

duguyue100, 2021-04-22:
Hi, @Kyrbouille, the story here is a bit messy. In the case that you want to write outputs into AEDAT2, then instead of setting `output_width` and `output_height`, you should add an optional argument `--dvs240` in the command.
This is because currently, AEDAT2 does not support arbitrary output width and height (I hope we can find the desired solution soon).

In terms of AEDAT2 to AEDAT4 conversion, since I use very little DV, I'm not sure what's the problem (can you also provide the conversion error log?). I knew that DV has a python binding that can decode AEDAT2, maybe they also have a saver there to save the events into AEDAT4, but in any case, you should try to consult DV's supporting team as well.
They have a dedicated Google group called `dv-users`, maybe you can ask a question there as well:https://groups.google.com/d/forum/dv-users/

tobidelbruck, 2021-04-22:
A contribution of AEDAT-4.0 output would be welcome. But we don't have 
time to write this.

The file format for all AEDAT format is very nicely documented in 
https://inivation.github.io/inivation-docs/Software%20user%20guides/AEDAT_file_formats.html

It would not be difficult in case of pure DVS output (as v2e currently 
produces). Supporting DAVIS emulation is a possible future target.

closed 2021-04-26 by tobidelbruck

tobidelbruck, 2021-07-28:
I just noticed this thread, sorry about long delay.

To convert from AEDAT2 to AEDAT4, you can use DV. It has a built in converter.
To convert from AEDAT4 to AEDAT2 for using DV recordings in jAER, see https://github.com/SensorsINI/aedat4to2