SensorsINI/v2e #17: The differences between the generated "clean" events and "noisy" events is not as expected

fleebeatles, 2021-04-23:
Hi, I wanted to investigate the influence of DVS noise. so I converted the provided sample video (**pendulum**) to events streams with _v2e_. I conducted two experiments with all the arguments same except the ones controlling the noise. The commands are as follows.

1. I expected this one generated clean events without noise.
```console
python v2e.py -i input/pendulum_trim.mp4  --overwrite  --auto_timestamp_resolution=False --timestamp_resolution=0.001  --dvs_exposure duration 0.005 --output_width=346 --output_height=260   --output_folder=output/pendulum_trim_wo_noise  --pos_thres=.2 --neg_thres=.2 --sigma_thres=0 --leak_rate_hz=0 --shot_noise_rate_hz=0 --cutoff_hz=0 --dvs_aedat2 pendulum.aedat   --dvs_params=None  --no_preview --ignore-gooey
```
2. I intended to generate noisy events.
```console
python v2e.py -i input/pendulum_trim.mp4  --overwrite  --auto_timestamp_resolution=False --timestamp_resolution=0.001  --dvs_exposure duration 0.005 --output_width=346 --output_height=260   --output_folder=output/pendulum_trim_noise1  --pos_thres=.2 --neg_thres=.2 --sigma_thres=0 --leak_rate_hz=0.01 --shot_noise_rate_hz=0.001 --cutoff_hz=0 --dvs_aedat2 pendulum.aedat   --dvs_params=None  --no_preview --ignore-gooey
```
Since the only difference of two commands is the `--leak_rate_hz` and `--shot_noise_rate_hz` parameter, I supposed that the generated stream of these two commands have many "events" in comman and the extra data generated by the 2nd command are all noisy events. However, the results showed that generated data were very much different and the event item generated with the 1<sup>st</sup> command can hardly find the counterpart in the 2<sup>nd</sup>one. 

Can anyone help me to figure out the problem? 

The following figure shows part of the generated data. The left column corresponds to the 1<sup>st</sup> command and the right 2<sup>nd</sup>.
[image]

tobidelbruck, 2021-04-23:
Been a while since we looked at pendulum data.... thanks for taking a look.

Where is the pendulum_trim.mp4 video? I don't think is is part of the 
project. I can only find our original pendulumXXX.avi files. In fact I 
don't see any input sample files in v2e anymore, not sure what happened 
to them.

fleebeatles, 2021-04-23:
> Been a while since we looked at pendulum data.... thanks for taking a look. Where is the pendulum_trim.mp4 video? I don't think is is part of the project. I can only find our original pendulumXXX.avi files. In fact I don't see any input sample files in v2e anymore, not sure what happened to them.

Thank you. I downloaded the *pendulum.avi* from the Google drive this repository shares. I clipped part of the video which is named *pendulum_trim.mp4*.

tobidelbruck, 2021-04-26:
BTW, including leak events will definitely change the non-noise events 
as well, because the leak affects the internal state of the Lmem of the 
pixel. Therefore the non-leak events will be affected too. See the model 
figure

closed 2021-04-26 by tobidelbruck