SensorsINI/v2e #19: avi_frame_rate parameter not working as expected

Franm99, 2021-05-03:
Hi.

I've been using this simulator (in my opinion, very useful and easy to use) to transform conventional video frames to DVS events. I'm trying to understand the usage of some parameters, and I think i'm not using the `avi_frame_rate` parameter correctly.

Reading its description, I supposed that the frame rate of the generated dvs-video.avi will be set by this parameter, but in my case, that file is always generated with 30fps frame rate. I've tried to set this parameter to different values (60fps, 120fps and 10fps), resulting in all cases in a video with the same frame rate.

Has I missunderstood how this parameter works? I expect that the resulting video will be shown at a different velocity because of this frame rate, but all videos look the same. How can I notice about any change produced by this parameter?

Thanks in advance.

tobidelbruck, 2021-05-03:
Thanks for the report, we will check it. You should be able to search 
(if running from project) to find the problem, maybe we forget to pass 
it to opencv aviwriter?

Franm99, 2021-05-04:
That's exactly the problem. The eventRenderer instance takes son arguments, but avi_frame_rate is set to 30 in the class constructor and never changed. The instance is created in v2e.py (line 430). Adding 'avi_frame_rate=args.avi_frame_rate' when creating the instance is enough to correct the issue. I checked it and worked.

tobidelbruck, 2021-05-04:
Thanks a lot Fran!  Yuhuang, could you push the fix, or should I?

duguyue100, 2021-05-04:
@tobidelbruck @Franm99 It's a good find. I've changed `v2e.py`. @Franm99 You can close the issue when it works to you.

closed 2021-05-04 by Franm99

tobidelbruck, 2021-05-04:
Great, thanks for finding this Fran. If you have feature suggestions, 
let us know. For example, a DAVIS simulator would be pretty easy. 
Computing the model parameters directly from bias currents could also be 
easy, at least if the DVS pixel circuit parameters are known.  We are 
considering the next useful feature to add is refractory period.