SensorsINI/v2e #20: Problem while using 'disable_slowmo' parameter

Franm99, 2021-05-06:
Hi.

I've run into an issue while testing the simulator if the 'disable_slowmo' parameter is used. I tried to run with the following set of parameters for one of the example videos:

```
--input=input/tennis.mov --output_folder=output/v2e-test --cutoff=0 --leak_rate=0 --shot=0 --dvs_exposure duration .03 --dvs_aedat2=v2e.aedat --output_width=346 --output_height=260 --batch=16 --disable_slomo --input_slowmotion_factor 1 --stop_time 0.5 --ignore-gooey --overwrite --no_preview
```

Getting the following output error:

```
...
Traceback (most recent call last):
  File "C:/Users/Usuario/Desktop/Sistemas Inteligentes UJI/Asignaturas/ReAViPeRo/v2e/v2e/v2e.py", line 706, in <module>
    main()
  File "C:/Users/Usuario/Desktop/Sistemas Inteligentes UJI/Asignaturas/ReAViPeRo/v2e/v2e/v2e.py", line 559, in main
    interpTimes, avgUpsamplingFactor = slomo.interpolate(
AttributeError: 'NoneType' object has no attribute 'interpolate'
```

Searching for a possible solution, I found out that the problem could reside on v2e.py (line 545):
```
# v2e.py - line 545:
if auto_timestamp_resolution or slowdown_factor != NO_SLOWDOWN:
    # interpolated frames are stored to tmpfolder as
    # 1.png, 2.png, etc

    logger.info(
        f'*** Stage 2/3: SloMo upsampling from '
        f'{source_frames_dir}')
    interpTimes, avgUpsamplingFactor = slomo.interpolate(
        source_frames_dir, interpFramesFolder,
        (output_width, output_height))
```

If 'disable_slomo' is used (setting slowdown_factor to NO_SLOWDOWN value, equals 1.0), the 'slomo' object is never built, so this statement should not occur.

I solved it by changing the 'or' operator in the if statement for an 'and' operator:

```
if auto_timestamp_resolution and slowdown_factor != NO_SLOWDOWN:
```

This way, if the user won't use the SlowMotion algorithm, maintaining the input video timestamp resolution, the else statement will occur:

```
else:
    logger.info(
        f'*** Stage 2/3:turning npy frame files to png '
        f'from {source_frames_dir}')
    interpFramesFilenames = []
    n = 0
    src_files = sorted(
        glob.glob("{}".format(source_frames_dir) + "/*.npy"))
    for frame_idx, src_file_path in tqdm(
            enumerate(src_files), desc='npy2png', unit='fr'):
        src_frame = np.load(src_file_path)
        tgt_file_path = os.path.join(
            interpFramesFolder, str(frame_idx) + ".png")
        interpFramesFilenames.append(tgt_file_path)
        n += 1
        cv2.imwrite(tgt_file_path, src_frame)
    interpTimes = np.array(range(n))
```

Is that a possible issue? Or maybe I missunderstood the use of the 'disable_slomo' parameter?

tobidelbruck, 2021-05-06:
Thank you Fran, you are great to track down all these bugs!

I'm sure it is some combo of the zillion options not tested by us.   We 
will check and push your fix.

BTW, you are running from fork/clone with latest code, correct? On 
master branch?

Franm99, 2021-05-07:
It's a pleasure for me! I'll notify you if I found more bugs.

Yes! I forked your repo, so I could add my own changes if neccesary for debugging and understanding how the simulator works.

I'll close this issue as it's been already resolved.

closed 2021-05-07 by Franm99