SensorsINI/v2e #21: --skip_video_output failed

GEN418, 2021-06-17:
When this parameter is added to the command line,  the procedure still produces video_orig.avi and video_slomo.avi.

duguyue100, 2021-06-17:
Hi, @GEN418 , this is a good find. I include this argument but not fully tested.

Now I fixed the bug and the argument will properly suppress all video outputs.

You can now pull and try again, let me know if it's not working for you.

Yuhuang.

GEN418, 2021-06-17:
Hi, @duguyue100. After pulling and running, I meet a mistake caused by the modification, so I try to fix it by changing 'is none' to 'is not none'  in v2ecore/slomo.py, the mistake and bug are gone. Thank you for your answer.
Gen.
[image]

duguyue100, 2021-06-17:
Definitely, I overkilled when I made the bug fixes, now it should be ok. thanks for reporting!

closed 2021-06-18 by GEN418