SensorsINI/v2e #22: AttributeError: 'str' object has no attribute 'appendEvents'

DanBigioi, 2021-07-23:
When I run the following command with the tennis.mov video as input:

`python v2e.py -i input/tennis.mov --overwrite --timestamp_resolution=.003 --auto_timestamp_resolution=False --dvs_exposure duration 0.005 --output_folder=output/tennis --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2 tennis.aedat --output_width=346 --output_height=260 --stop_time=3 --cutoff_hz=15`

 I get the following error: 

```
INFO:__main__:*** Stage 3/3: emulating DVS events from 1551 frames

dvs:   0%|          | 0/1551 [00:00<?, ?fr/s]
dvs:   0%|          | 2/1551 [00:00<01:52, 13.74fr/s]
dvs:   0%|          | 2/1551 [00:00<07:24,  3.48fr/s]
Traceback (most recent call last):
  File "v2e.py", line 719, in <module>
    main()
  File "v2e.py", line 658, in main
    fr, interpTimes[i])
  File "C:\Users\dabigioi\Documents\v2e\v2ecore\emulator.py", line 624, in generate_events
    self.dvs_aedat2.appendEvents(events)
AttributeError: 'str' object has no attribute 'appendEvents'
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "C:\Users\dabigioi\Documents\v2e\v2ecore\emulator.py", line 217, in cleanup
    self.dvs_aedat2.close()
AttributeError: 'str' object has no attribute 'close'
```

Any clue what is happening?

tobidelbruck, 2021-07-23:
No idea right now, we will try to reproduce it.

The exception is somehow caused by a string object called aedat2... we 
have to try your exact command line.

Did you pull latest master?

Did you try running with no arguments? i.e. just "v2e"?

tobidelbruck, 2021-07-23:
By the way, could you check if the pip install -e . works for you? Does 
v2e get installed as script to your conda shell?

DanBigioi, 2021-07-23:
Hi Tobi, 

Yup im running on the latest branch, and pip install =e . works as well for me. Running with no arguments just v2e gives me the same error unfortunately. 

I was running into the same issue described here: https://github.com/SensorsINI/v2e/issues/20
I followed that fix and now this error.

tobidelbruck, 2021-07-24:
Yuhuang, will you look at this?

duguyue100, 2021-07-26:
@tobidelbruck I will look at this.

@DanBigioi From what I can see here, maybe the error is due to the AEDAT writer doesn't support arbitrary output dimensions. I will get back to you ASAP.

nuaayxy, 2021-07-28:
> @tobidelbruck I will look at this.
> 
> @DanBigioi From what I can see here, maybe the error is due to the AEDAT writer doesn't support arbitrary output dimensions. I will get back to you ASAP. Meanwhile, try to append some arguments such as `--davis346`.

seems like this only works with --dvs346 or lower. anything else breaks at :
Traceback (most recent call last):
  File "v2e.py", line 719, in <module>
    main()
  File "v2e.py", line 658, in main
    fr, interpTimes[i])
  File "C:\Users\dabigioi\Documents\v2e\v2ecore\emulator.py", line 624, in generate_events
    self.dvs_aedat2.appendEvents(events)
AttributeError: 'str' object has no attribute 'appendEvents'
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "C:\Users\dabigioi\Documents\v2e\v2ecore\emulator.py", line 217, in cleanup
    self.dvs_aedat2.close()
AttributeError: 'str' object has no attribute 'close'

tobidelbruck, 2021-07-28:
Yes, that's exactly it.  Look in 
https://github.com/SensorsINI/v2e/blob/296c66de4dbfcda07f5eaa9822de92661f01f6ab/v2ecore/output/aedat2_output.py

There are only 2 cameras supported now, Davis346 and Davis240.

What do you need?

The AEDAT output must match some AEChip type in jAER since jAER depends 
on this AEChip software object model of the hardware to understand any 
AEDAT file input.

tobidelbruck, 2021-07-28:
> Hi Tobi,
> 
> Yup im running on the latest branch, and pip install =e . works as well for me. Running with no arguments just v2e gives me the same error unfortunately.
> 
> I was running into the same issue described here: #20
> I followed that fix and now this error.

By the way, the better way to install using setup.py is "python setup.py develop". The pip install does not seem to work properly sometimes, e.g. in google colab. We don't know why. The paths are not setup correctly.

tobidelbruck, 2021-07-28:
I just tried to reproduce the original error by running exactly the command line
````
python v2e.py -i input/tennis.mov --overwrite --timestamp_resolution=.003 --auto_timestamp_resolution=False --dvs_exposure duration 0.005 --output_folder=output/tennis --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2 tennis.aedat --output_width=346 --output_height=260 --stop_time=3 --cutoff_hz=15
````
I did not get the exception. The conversion completed successfully.
Could it be that you missed seeing some previous error in output log?

One recent error I've seen is related to "rounding". Since v2e now uses torch to accelerate DVS modeling, it is necessary to install the very latest pytorch stuff.....  

It's a bit of a pain. I managed to get it working on windows by using conda installs of the packages following the requirements in the pip setup.py file.

You may want to simply try the google colab option if your conversions take <40m (or if you have colab pro). See the "Open in colab" button at top of README in v2e github.

tobidelbruck, 2021-07-28:
[image]

Here is the output log
````
tobid@tobi-joule-amd MINGW64 /f/tobi/Dropbox (Personal)/GitHub/SensorsINI/v2e (master)
$ python v2e.py -i input/tennis.mov --overwrite --timestamp_resolution=.003 --auto_timestamp_resolution=False --dvs_exposure duration 0.005 --output_folder=output/tennis --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2 tennis.aedat --
output_width=346 --output_height=260 --stop_time=3 --cutoff_hz=15
←[1;31mWARNING←[1;0m:__main__:No module named 'gooey': Gooey GUI builder not available, will use command line arguments.
Install with 'pip install Gooey'. See README
INFO:__main__:name 'Gooey' is not defined: Gooey package GUI not available, using command line arguments.
You can try to install with "pip install Gooey"
INFO:__main__:Command line:
 v2e.py -i input/tennis.mov --overwrite --timestamp_resolution=.003 --auto_timestamp_resolution=False --dvs_exposure duration 0.005 --output_folder=output/tennis --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2 tennis.aedat --output_w
idth=346 --output_height=260 --stop_time=3 --cutoff_hz=15
INFO:v2ecore.v2e_utils:using output folder F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\output\tennis
INFO:__main__:output_in_place==False so made output_folder=F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\output\tennis
INFO:v2ecore.v2e_args:DVS frame expsosure mode ExposureMode.DURATION: frame rate 200.0
INFO:v2ecore.v2e_args:arguments:
output_folder:  F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\output\tennis
avi_frame_rate: 30
output_in_place:        True
overwrite:      True
unique_output_folder:   True
auto_timestamp_resolution:      False
timestamp_resolution:   0.003
output_height:  260
output_width:   346
dvs_params:     None
pos_thres:      0.15
neg_thres:      0.15
sigma_thres:    0.03
cutoff_hz:      15.0
leak_rate_hz:   0.01
shot_noise_rate_hz:     0.001
leak_jitter_fraction:   0.1
noise_rate_cov_decades: 0.1
refractory_period:      0.0005
dvs_emulator_seed:      0
show_dvs_model_state:   None
dvs128: False
dvs240: False
dvs346: False
dvs640: False
dvs1024:        False
disable_slomo:  False
slomo_model:    F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\input\SuperSloMo39.ckpt
batch_size:     8
vid_orig:       video_orig.avi
vid_slomo:      video_slomo.avi
slomo_stats_plot:       False
input:  F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\input\tennis.mov
input_frame_rate:       None
input_slowmotion_factor:        1.0
start_time:     None
stop_time:      3.0
synthetic_input:        None
dvs_exposure:   ['duration', '0.005']
dvs_vid:        dvs-video.avi
dvs_vid_full_scale:     2
skip_video_output:      False
no_preview:     False
davis_output:   False
dvs_h5: None
dvs_aedat2:     tennis.aedat
dvs_text:       v2e-dvs-events.txt

INFO:__main__:opening video input file F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\input\tennis.mov
INFO:__main__:--auto_timestamp_resolution is False, srcFps=59.94007280921804Hz input_slowmotion_factor=1.0, real src FPS=59.94007280921804Hz, srcFrameIntervalS=16.68ms, timestamp_resolution=3ms, so SuperSloMo will use slowdown_factor=6 and have slomoTimestampRe
solutionS=2.78ms
INFO:__main__:Lowpass cutoff is f_3dB=15Hz with tau=10.61ms and with sample rate fs=359.64Hz (sample interval dt=2.78ms),
[... 33 lines cut ...]
Average upsampling factor=  6.0
INFO:__main__:SloMo average upsampling factor= 6.00; average DVS timestamp resolution=2.78ms
←[1;31mWARNING←[1;0m:__main__:Using auto_timestamp_resolution. checking if cutoff hz is ok given samplee rate 359.6404368553082
INFO:__main__:Lowpass cutoff is f_3dB=15Hz with tau=10.61ms and with sample rate fs=359.64Hz (sample interval dt=2.78ms),
It has IIR mixing factor eps=0.262 which is OK because it is less than recommended maxeps= 0.3
INFO:__main__:*** Stage 3/3: emulating DVS events from 1074 frames
dvs:   0%|▏                                                                                                                                                                                                                        | 1/1074 [00:00<03:19,  5.38fr/s]I
NFO:v2ecore.renderer:opening DVS video output file F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\output\tennis\dvs-video.avi
INFO:v2ecore.renderer:opening DVS frame times file F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\output\tennis\dvs-video-frame_times.txt
dvs: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1074/1074 [01:08<00:00, 15.69fr/s]
INFO:v2ecore.renderer:Closing DVS video output file after writing 595 frames
INFO:v2ecore.output.aedat2_output:Closing F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\output\tennis\tennis.aedat after writing 3M events (1.52M on, 1.48M off)
INFO:v2ecore.output.ae_text_output:Closing F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\output\tennis\v2e-dvs-events.txt after writing 3M events
INFO:v2ecore.slomo:closing original video AVI after writing 180 frames
INFO:v2ecore.slomo:closing slomo video AVI after writing 1074 frames
INFO:__main__:done processing 180 frames in 116.15s (1.55fr/s)
 see output folder F:\tobi\Dropbox (Personal)\GitHub\SensorsINI\v2e\output\tennis
INFO:__main__:generated total 3M events (1.52M on, 1.48M off)
INFO:__main__:avg event rate 1MHz (507.67kHz on, 495.53kHz off)
INFO:v2ecore.renderer:Closing DVS video output file after writing 595 frames
INFO:v2ecore.slomo:closing original video AVI after writing 180 frames
INFO:v2ecore.slomo:closing slomo video AVI after writing 1074 frames

tobid@tobi-joule-amd MINGW64 /f/tobi/Dropbox (Personal)/GitHub/SensorsINI/v2e (master)
````

nuaayxy, 2021-07-28:
> Yes, that's exactly it.  Look in https://github.com/SensorsINI/v2e/blob/296c66de4dbfcda07f5eaa9822de92661f01f6ab/v2ecore/output/aedat2_output.py There are only 2 cameras supported now, Davis346 and [...]

What about when you feed 640x480 or high resolution images/video? I tried that and if set use dvs346 it still generate something. Is it just downscaled events in coordinates?? the slomo can generate the same resolution though. 
Would really be nice if it can support at least 640x480 which is kinda very general use case of camera resolutions.

tobidelbruck, 2021-08-02:
I finally had some time and added a bunch of sanity checking for valid 
v2e inputs and options for input folders and AEDAT output width and 
height options.

Please pull changes and give it a try?

BTW, if you do "python setup.py develop" you can pull changes and get 
immediate fixes.

DanBigioi, 2021-08-18:
Sorry for the delay in response, I went off on vacation the last couple of weeks 😅 I reinstalled everything pulling all new changes and conda installed all packages like you mentioned in a response above, its working! I think we can close this issue now 😃 
One quick question though, when you guys mentioned that only davis346 and davis240 is supported, does that mean that only images/videos of size 346x260 and 240x180 work in v2e? (Sorry if its a bad question, Im pretty new to events 😅 ) What is the procedure to add support for new resolutions?

duguyue100, 2021-08-18:
@DanBigioi, not necessarily, when you pass `--davis346` or `--davis240` option, the input frame will be resized to this chip size when necessary. In other cases, the frame will either keep its original size or resize to a preset dimension via the arguments `--output_height` and `--output_width`. I will close this issue now and we can keep discussing in this thread.

closed 2021-08-18 by duguyue100

DanBigioi, 2021-08-18:
> @DanBigioi, not necessarily, when you pass `--davis346` or `--davis240` option, the input frame will be resized to this chip size when necessary. In other cases, the frame will either keep its [...]

Ah I see. Under what conditions will the frame get resized to the chip size? And are there any downsides to using a custom size over the original 346/240 ones?