SensorsINI/v2e #23: AttributeError: 'NoneType' object has no attribute 'interpolate'

AmoghTiwari, 2021-09-06:
Hi, 

I was trying to generate simulated events with the slowmo step disabled (i.e. by using the ``` --disable_slomo ``` argument. However, I run into an error when I do that. The same video works fine when I don't use the ``` --disable_slomo ``` argument. 

I want to disable slomo, because I want to do some custom experiments with the slowmo parts, and hence, I had already upsampled the video before sending it to the simulator.

The error message and some lines preceeding it are given below:

``` rgb2luma: 100%|████████████████████████████████████████████████████████████████████████████████████████████| 722/722 [00:06<00:00, 115.32fr/s]
INFO:__main__:*** Stage 2/3: SloMo upsampling from /tmp/tmp9h82o98d
Traceback (most recent call last):
  File "/home/amoghtiwari/v2e/v2e.py", line 805, in <module>
    main()
  File "/home/amoghtiwari/v2e/v2e.py", line 657, in main
    interpTimes, avgUpsamplingFactor = slomo.interpolate(
AttributeError: 'NoneType' object has no attribute 'interpolate'
INFO:v2ecore.output.aedat2_output:Closing /home/amoghtiwari/v2e/output/airboard_1_30_my_slomo/tennis.aedat after writing 0 events (0 on, 0 off) ```

AmoghTiwari, 2021-09-10:
Update: I converted the video to mp4 online, and it worked then. But it still doesn't work with mkv files

tobidelbruck, 2022-05-18:
I guess it is  a problem with cv2 video reader....closing this issue.

closed 2022-05-18 by tobidelbruck