SensorsINI/v2e #25: Problem with v2e gooey GUI

angelsb008, 2021-11-30:
When I ran the instruction python **_-m v2e_** in the console, gui doesn't appear, only appear a window requesting the file to open, then I chose de file and appear the present error: 

(v2e) PS C:\Users\laloh\v2e> python -m v2e

INFO:__main__:Use --ignore-gooey to disable GUI and run with command line arguments
←[1;31mWARNING←[1;0m:__main__:Either output_width is None or output_height is None,or both. Setting both of them to None.Actualy dimension will be set automatically.
INFO:v2ecore.v2e_utils:using output folder C:\Users\laloh\v2e\v2e-output-1
INFO:__main__:output_in_place==False so made output_folder=C:\Users\laloh\v2e\v2e-output-1
INFO:v2ecore.v2e_args:arguments:
auto_timestamp_resolution:      True
avi_frame_rate: 30
batch_size:     8
crop:   None
cutoff_hz:      300
davis_output:   False
disable_slomo:  False
dvs1024:        False
dvs128: False
dvs240: False
dvs346: False
dvs640: False
dvs_aedat2:     v2e-dvs-events.aedat
dvs_emulator_seed:      0
dvs_exposure:   ['duration', '0.01']
dvs_h5: None
dvs_params:     None
dvs_text:       None
dvs_vid:        dvs-video.avi
dvs_vid_full_scale:     2
input:  None
input_frame_rate:       None
input_slowmotion_factor:        1.0
leak_jitter_fraction:   0.1
leak_rate_hz:   0.01
neg_thres:      0.2
no_preview:     False
noise_rate_cov_decades: 0.1
output_folder:  C:\Users\laloh\v2e\v2e-output
output_height:  None
output_in_place:        True
output_width:   None
overwrite:      False
pos_thres:      0.2
refractory_period:      0.0005
shot_noise_rate_hz:     0.001
show_dvs_model_state:   None
sigma_thres:    0.03
skip_video_output:      False
slomo_model:    C:\Users\laloh\v2e\input\SuperSloMo39.ckpt
slomo_stats_plot:       False
start_time:     None
stop_time:      None
synthetic_input:        None
timestamp_resolution:   None
unique_output_folder:   True
vid_orig:       video_orig.avi
vid_slomo:      video_slomo.avi

INFO:__main__:Command line:
 C:\Users\laloh\v2e\v2e.py
INFO:v2ecore.v2e_args:DVS frame expsosure mode ExposureMode.DURATION: frame rate 100.0
INFO:__main__:opening video input file C:/Users/laloh/v2e/input/tennis.mov
INFO:__main__:--auto_timestamp_resolution=True and timestamp_resolution is not set: source video will be automatically upsampled to limit maximum interframe motion to 1 pixel
←[1;31mWARNING←[1;0m:v2ecore.slomo:CUDA not available, will be slow :-(
INFO:v2ecore.slomo:Using auto_upsample and upsampling_factor; setting minimum upsampling to 1
INFO:v2ecore.slomo:using automatic upsampling mode
INFO:__main__:Source video C:/Users/laloh/v2e/input/tennis.mov has total 1551 frames with total duration 25.86s.
Source video is 59.94fps with slowmotion_factor 1 (frame interval 16.68ms),
Will convert 1551 frames 0 to 1550
(From 0.0s to 25.859161114693027s, duration 25.859161114693027s)
INFO:__main__:v2e DVS video will have constant-duration frames
at 100fps (accumulation time 10ms),
DVS video will have 2585 frames with duration 25.85s and playback duration 86.17s

INFO:v2ecore.emulator:ON/OFF log_e temporal contrast thresholds: 0.2 / 0.2 +/- 0.03
INFO:v2ecore.emulator:opening AEDAT-2.0 output file C:\Users\laloh\v2e\v2e-output-1\v2e-dvs-events.aedat
←[1;41mERROR←[1;0m:v2ecore.emulator:Output file exception "AEDAT-2.0 output width=None height=None not supported; add your camera to v2ecore.output.aedat2_output or use one of the predefined DVS cameras, e.g. --dvs346 or --dvs240 that have sizes ((346, 260), (240, 180))" (maybe you need to specify a supported DVS camera type?)
Traceback (most recent call last):
  File "C:\Users\laloh\anaconda3\envs\v2e\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\laloh\anaconda3\envs\v2e\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\laloh\v2e\v2e.py", line 805, in <module>
    main()
  File "C:\Users\laloh\v2e\v2e.py", line 462, in main
    emulator = EventEmulator(
  File "C:\Users\laloh\v2e\v2ecore\emulator.py", line 181, in __init__
    raise e
  File "C:\Users\laloh\v2e\v2ecore\emulator.py", line 171, in __init__
    self.dvs_aedat2 = AEDat2Output(
  File "C:\Users\laloh\v2e\v2ecore\output\aedat2_output.py", line 59, in __init__
    raise ValueError(f'AEDAT-2.0 output width={output_width} height={output_height} not supported; add your camera to {__name__} or use one of the predefined DVS cameras, e.g. --dvs346 or --dvs240 that have sizes {self.SUPPORTED_SIZES}')
ValueError: AEDAT-2.0 output width=None height=None not supported; add your camera to v2ecore.output.aedat2_output or use one of the predefined DVS cameras, e.g. --dvs346 or --dvs240 that have sizes ((346, 260), (240, 180))

tobidelbruck, 2021-11-30:
The exception is caused by specifying AEDAT-2.0 output but not 
specifying a supported output array size, i.e. 128x128, 240x180, or 
346x260. The fix is exactly specified by the error message. Or am I 
missing something?

> ←[1;41mERROR←[1;0m:v2ecore.emulator:Output file exception "AEDAT-2.0 
> output width=None height=None not supported; add your camera to 
> v2ecore.output.aedat2_output or use one of the predefined DVS cameras, 
> e.g. --dvs346 or --dvs240 that have sizes ((346, 260), (240, 180))" 
> (maybe you need to specify a supported DVS camera type?)
> Traceback (most recent call last):
> File "C:\Users\laloh\anaconda3\envs\v2e\lib\runpy.py", line 197, in 
> _run_module_as_main
> return _run_code(code, main_globals, None,
> File "C:\Users\laloh\anaconda3\envs\v2e\lib\runpy.py", line 87, in 
> _run_code
> exec(code, run_globals)
> File "C:\Users\laloh\v2e\v2e.py", line 805, in
> main()
> File "C:\Users\laloh\v2e\v2e.py", line 462, in main
> emulator = EventEmulator(
> File "C:\Users\laloh\v2e\v2ecore\emulator.py", line 181, in *init*
> raise e
> File "C:\Users\laloh\v2e\v2ecore\emulator.py", line 171, in *init*
> self.dvs_aedat2 = AEDat2Output(
> File "C:\Users\laloh\v2e\v2ecore\output\aedat2_output.py", line 59, in 
> *init*
> raise ValueError(f'AEDAT-2.0 output width={output_width} 
> height={output_height} not supported; add your camera to {*name*} or 
> use one of the predefined DVS cameras, e.g. --dvs346 or --dvs240 that 
> have sizes {self.SUPPORTED_SIZES}')
> ValueError: AEDAT-2.0 output width=None height=None not supported; add 
> your camera to v2ecore.output.aedat2_output or use one of the 
> predefined DVS cameras, e.g. --dvs346 or --dvs240 that have sizes 
> ((346, 260), (240, 180))
>
> or Android 
>
>
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage", 
> "https://github.com/SensorsINI/v2e/issues/25", "url": 
> "https://github.com/SensorsINI/v2e/issues/25", "name": "View Issue" }, 
> "Organization", "name": "GitHub", "url": "https://github.com" } } ]

Aces2017, 2022-01-26:
> The exception is caused by specifying AEDAT-2.0 output but not specifying a supported output array size, i.e. 128x128, 240x180, or 346x260. The fix is exactly specified by the error message. Or am I missing something?
> […](#)
> -- Tobi Delbruck, Sensors Group:http://sensors.ini.uzh.ch Inst. for Neuroinformatics, UZH-ETH Zurich:http://www.ini.uzh.ch Building 55, Room G84, Winterthurerstr. 190, CH-8057 Zurich, Switzerland Tel lab +41 44 635 3038, Swiss cell +41 76 629 1500 (whatsapp), USA cell +1 626 510 2646, skype tobidelbruck

I had the same problem as @angelsb008 , the main problem is NO GUI appear. Instead, i changed the default value of output width and output height to 346×260, and got only simulated event video and aedat document, without txt which contained x,y,t,p. Could you please give more indications about how to get the GUI when running v2e.py? Thanks very much!

tobidelbruck, 2022-01-26:
Do you mean the gooey GUI?  You want the GUI to appear that allows 
setting options and selecting file? I believe we removed the gooey code 
since it caused so many problems with users running on linux, since 
gooey requires a lot of native code components and is quite difficult to 
get working for most linux systems (the pre-built packages are a bit 
flaky).

I might not be remembering correctly. Sorry for the confusion about 
this... if really useful, we could add gooey back in. Another good 
option for flexible usage is the notebook version of v2e. Please comment.

Aces2017, 2022-01-26:
> Do you mean the gooey GUI?  You want the GUI to appear that allows setting options and selecting file? I believe we removed the gooey code since it caused so many problems with users running on [...]

Thanks a lot for explanations about gooey, and I'll try the notebook version. Would it be possible to update a version containing gooey code, maybe which is friendly to windows users?Best whishes!

duguyue100, 2022-01-26:
@Aces2017 , actually for Windows, the Gooey should still be there if you have the package installed `pip install Gooey`.

From your previous post, I think you are looking for visualization during event simulation? If you have enabled one of the following options: 1. `--skip_video_output`, 2. `--no_preview`, it's unlikely you will get any OpenCV-driven previews.
Usually, for faster processing, I just disable all the previews and write my custom visualization code in the end.

For simple debugging, as Tobi suggested, you could use our Google Colab notebook to play around. Here is the link: https://colab.research.google.com/drive/1czx-GJnx-UkhFVBbfoACLVZs8cYlcr_M?usp=sharing

Aces2017, 2022-01-26:
> @Aces2017 , actually for Windows, the Gooey should still be there if you have the package installed `pip install Gooey`.
>
> From your previous post, I think you are looking for visualization during [...]

All right, thanks a lot!

tobidelbruck, 2022-01-28:
Since you guys are active users, maybe you could suggest what feature is 
most missing from v2e now for future enhancement? What is most annoying 
now? (Besides speed of course; not too much we can do about this at 
present except to switch to a different flow inference network)

closed 2022-05-18 by tobidelbruck