SensorsINI/v2e #28: Input RGB frames, how to get timestamps of RGB after V2E

Ginobili-20, 2021-12-09:
A fantastic tool! 
Sorry to bother u. When the input files are raw RGB frames, the input_frame_rate is set as 240. How can I get the exact RGB timestamp after the simulation?
THX a lot

duguyue100, 2021-12-09:
Hi, @Ginobili-20,

Recording the frame and timestamp is supported in the option `--davis_output`, you can check it here: https://github.com/SensorsINI/v2e/blob/master/v2ecore/v2e_args.py#L350

This option also records the corresponding event index for the respective frame.

However, when writing the frames, it makes the event emulation slow because of constant frame writing to the disk.

The following code is when the emulator wrote frame timestamps: https://github.com/SensorsINI/v2e/blob/6991adb89a1ec89866cc36be1126888fd6934a51/v2ecore/emulator.py#L195-L201

If you only care about frame timestamps, maybe the most convenient way, for now, is saving the variable `interpTimes` before this line: https://github.com/SensorsINI/v2e/blob/master/v2e.py#L732

write one or two lines to save the timestamps into a `npy` file and you are good to go.

Regards,
Yuhuang.

Ginobili-20, 2021-12-09:
Wonderful and detailed solutions!!! THX for your immediate reply!!! 
Best Wishes

closed 2021-12-09 by Ginobili-20