SensorsINI/v2e #30: The CUDA memory kept growing when transfer videos into events recursively.

gavinatthu, 2022-01-07:
Here I'm trying to transfer series of videos into events through v2e method. I recursively use the `def main() ` function in v2e.py to transfer around 1000 short videos into events. But the CUDA memory kept growing till "RuntimeError: CUDA error: out of memory".
My code:
`from v2e import *
def main():
    video_ls = os.listdir(input_path)
    for i in video_ls:
        video_path = os.path.join(input_path, i)
        event_path = os.path.join(output_path, i[:-4]+".npz")
        if os.path.exists(event_path):
            continue
        run_v2e(video_path)`
where `def run_v2e()` is `def main() ` in your v2e.py
Then I got growing CUDA memory from 2500MiB in the beginning till out of memory.
Here's the beginning in GPU2:
[image]
Here's after some runs:
[image]
The memory grow up to 6000MiB due to some reason.

tobidelbruck, 2022-01-07:
> image
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
> "https://github.com/SensorsINI/v2e/issues/30", "name": "View Issue" },
> "Organization", "name": "GitHub", "url": "https://github.com" } } ]

Interesting, no idea where is the leak. We never tried it like this.

Why not just script it via shell? I guess they are synthetic vids?  No 
idea how to force garbage collection on GPU...

duguyue100, 2022-01-07:
@gavinatthu,

I'm also not sure why this is caused, will try to find out. In the meantime, I suggest that your use `subprocess` in Python instead of direct import when converting a batch of videos. The basic idea is to use a master script to run each conversion command individually. A concrete example is here:

https://github.com/SensorsINI/v2e_exps_public/blob/main/scripts/datasets/ncaltech101_data_maker.py

Take particular attention to these lines: https://github.com/SensorsINI/v2e_exps_public/blob/main/scripts/datasets/ncaltech101_data_maker.py#L116-L137

gavinatthu, 2022-01-07:
Thanks Tobi and Yuhuang, it work in my situation.

For the problem, I tried `torch.cuda.empty_cache()` after each generation, but it won't work.

> @gavinatthu,
>
> I'm also not sure why this is caused, will try to find out. In the meantime, I suggest that your use `subprocess` in Python instead of direct import when converting a batch of videos. [...]

closed 2022-01-12 by gavinatthu