SensorsINI/v2e #35: evet time stamps do not match with video

oshima-yoppi, 2022-04-28:
Thank you for developing such a brilliant tool!!

The timestamps do not match the original video, even though it is set to no slow motion. 
Even though the video is at 100 fps, the event timestamps are not every 0.01s.
I would like to know why this is so.

The command I used is as follows.
`python v2e.py -i input\0.avi  --output_folder=output/tennis --dvs_text DVS_TEXT --dvs_exposure duration 0.01 --overwrite --timestamp_resolution=0.01 --auto_timestamp_resolution=False  --output_folder=output/tennis --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2 tennis.aedat --output_width=240 --output_height=180  --cutoff_hz=15 --disable_slomo --timestamp_resolution=0.01 --input_slowmotion_factor 1 `

tobidelbruck, 2022-04-28:
With --disable_slomo, v2e should just use the frame times for DVS event 
timestamps, unless there are more than 1 event per input frame. We had 
to choose some scheme for spreading these events over time between 
frames. The unfortunate side effect is a regular quantization of the 
timestamps with clumps of more events at the frame times. These 
additional events are spread over the the time between frames.  I.e., 
pixels that make only 1 event will have 10ms timestamp. For each frame 
difference, v2e finds the pixel(s) with the most number of events, e.g. 
9 events. Then those pixels will have 1ms timestamps and other pixels 
with smaller change will make fewer events, down to pixels with only a 
single event.  You will notice that more pixels will make timestamps 
with 10ms quantization since they make only 1 event. This explanation is 
probably not so clear, but try to visualize the output events with 3d 
spacetime view and you will notice pyramids of events.   Does this 
explanation help clarify what is going on?

tobidelbruck, 2022-05-18:
I have added clarifying video to README.md.

closed 2022-05-18 by tobidelbruck