SensorsINI/v2e #37: Incorrect x,y order in ae_text_output.appendEvents()

abdulmohsij, 2022-05-18:
EventEmulator.generate_events() returns events as an np.ndarray with each row being: [timestamp, y cordinate, x cordinate, sign of event]

These events are then passed into the text output:
self.dvs_text.appendEvents(events)

In appendEvents:
x = events[:, 1].astype(np.int32)
...
y = events[:, 2].astype(np.int32)

This is incorrect.
It should be:
x = events[:, 2].astype(np.int32)
...
y = events[:, 1].astype(np.int32)

tobidelbruck, 2022-05-18:
Thanks, I agree. This could mess up some users who have adapted to this order.  I have pushed this change to master and will make additional pushes to document if needed.

closed 2022-05-18 by tobidelbruck

tobidelbruck, 2022-10-05:
This push was wrong, it was based on incorrect docstring. See #48 for continued thread. Order in text output is again correct.