SensorsINI/v2e #38: !$final_v2e_command doesn't work

beratersari, 2022-05-30:
[image]
after the v1.6.0 this command doesn't work. How can I solve this problem?

tobidelbruck, 2022-05-30:
It should work after running "pip install -e setup.py", but I will check 
that the v2e script is correctly installed still, hopefully tomorrow. 
Otherwise you can always run "python v2e.py" if your paths are setup 
correctly.

beratersari, 2022-05-31:
I am working on Google Colab. 

The link is provided here.

Using v2e in Google CoLab
If you don't want to install, try [opening v2e in google colab](https://colab.research.google.com/drive/1czx-GJnx-UkhFVBbfoACLVZs8cYlcr_M?usp=sharing).

It worked before 20 days ago. However, it is not working now.

Thanks

feihigh24, 2022-08-17:
I have the same problem

tobidelbruck, 2022-08-17:
I think this is something Yuhuang has to fix. I'm not sure how he did this colab notebook. Maybe it auto-deleted? Yuhuang? Thanks!

tobidelbruck, 2022-08-17:
The link works for me, at least I came to the notebook. Are you signed into google? Or maybe it now is blocked in your locale?

duguyue100, 2022-08-17:
Looks like a persistent  error then, I will take a look later and let you know.

tobidelbruck, 2022-08-17:
I don't see what is the error. If I open the link "try [opening v2e in google colab](https://colab.research.google.com/drive/1czx-GJnx-UkhFVBbfoACLVZs8cYlcr_M?usp=sharing)." it works for me. At least it opens the notebook in colab.

feihigh24, 2022-08-18:
Maybe it's the connection problem of Google colab? Now it can work normally

tobidelbruck, 2022-10-11:
I think maybe Yuhuang can fix this easily.