SensorsINI/v2e #40: final_v2e_command gives error

gwgknudayanga, 2022-08-02:
I have tried to run the final_v2e_command command for the same video given on google collab. But It always give the following error.
Could you please help to resolve this?

WARNING:v2e:Either output_width is None or output_height is None,or both. Setting both of them to None.Actualy dimension will be set automatically.
Traceback (most recent call last):
  File "/usr/local/bin/v2e", line 8, in <module>
    sys.exit(main())
  File "/usr/local/bin/v2e.py", line 156, in main
    num_pixels=output_width*output_height
TypeError: unsupported operand type(s) for *: 'NoneType' and 'NoneType'

Thanks and Rgds,
Udayanga

tobidelbruck, 2022-08-02:
What is the command? It sounds like output size is not set. You can use 
--dvs346 for example to output Davis346 346x260 output.

gwgknudayanga, 2022-08-02:
Dear Tobi,

Thanks  for replying. This is the command that was used and it is the default command which is generated as we follow the steps in the the collab. 

v2e -i /content/v2e_tutorial_video.avi -o /content/v2e-output --overwrite --unique_output_folder false --davis_output --dvs_h5 events.h5 --dvs_aedat2 None --dvs_text None --no_preview --dvs_exposure duration .033 --input_frame_rate 30 --input_slowmotion_factor 1 --disable_slomo --auto_timestamp_resolution false --pos_thres 0.2 --neg_thres 0.2 --sigma_thres 0.03 --cutoff_hz 30 --leak_rate_hz 0.1 --shot_noise_rate_hz 5'

* However for the successful installation of v2e in the collab, i did not run followings before checking out the code for v2e to the collab as it gave numpy version mismatches (as some of the followings require numpy-1.2.1 as v2e requires numpy-1.2.0
%pip install numba
%pip install engineering-notation
%pip install opencv-contrib-python
%pip install argcomplete.

gwgknudayanga, 2022-08-03:
Hi,

Replacing --davis_output  option in the previously mentioned command with --dvs346 solved the problem. Thanks Tobi.

gwgknudayanga, 2022-08-03:
By the way, if there is any user guide or something like that regarding the command and command options please share as in the collab it may not contain some modified options like '--dvs346' while instead having  '-davis_output' options.

Thanks !

tobidelbruck, 2022-08-03:
Thanks. The user guide is the README.md 
https://github.com/SensorsINI/v2e/blob/master/README.md and of course 
the v2e paper which describes the model.

But looking at the parameters, I think we could stand to slightly 
reorganize them. The /output_width/ and /output_height/ are not really 
DVS model parameters. They are more like output options for determining 
the number of pixels we simulate and also the AER data format for 
supported sensors.

I will take a look once again at these options to see how to group them 
better.

In practice, for a paper like our CSDVS paper 
https://sites.google.com/view/csdvs/home we used a shell script with the 
key parameters set from the bash script, e.g.

#!/bin/bash
# scripts to reproduce results in 2022 CSDVS ICIP paper Special session 
20.11 ‘Neuromorphic and perception-based image acquisition and analysis”
# UTILITY AND FEASIBILITY OF A CENTER SURROUND EVENT CAMERA

# spots
# parame for CSDVS
lambda=10
tau=.5
dt=.0001
seq=spots
contrast=2
thr=.2
sig_thr=0.02
cutoff=100
refr=1e-3
exp="duration .01"

#v2e --output_folder=output/$seq-dvs --pos_thr=$thr --neg_thr=$thr 
--sigma_thr=$sig_thr --cutoff_hz=$cutoff --refractory_period=$refr 
--leak_rate_hz=0 --shot_noise_rate_hz=0  --vid_orig=inten --dvs_vid=dvs 
--unique_output --dvs_aedat2=dvs --output_width=346 --output_height=260 
--batch=1 --disable_slomo --synthetic_input=scripts.spots --dt=$dt 
--freq=20 --contrast=$contrast --dvs_exposure $exp 
--shot_noise_rate_hz=1 --show_dvs_model_state all --save_dvs_model_state
#v2e --output_folder=output/$seq-csdvs-l$lambda-t$tau 
--cs_lambda_pixels=$lambda --cs_tau_p_ms=$tau  --pos_thr=$thr 
--neg_thr=$thr --sigma_thr=$sig_thr --cutoff_hz=$cutoff 
--refractory_period=$refr --leak_rate_hz=0 --shot_noise_rate_hz=0  
--vid_orig=inten --dvs_vid=csdvs --unique_output --dvs_aedat2=csdvs 
--output_width=346 --output_height=260 --batch=1 --disable_slomo 
--synthetic_input=scripts.spots --dt=$dt --freq=20 --contrast=$contrast 
--dvs_exposure $exp --shot_noise_rate_hz=1 --show_dvs_model_state all 
--save_dvs_model_state

# gradients
seq=gradients

#v2e  --output_folder output/$seq-dvs --pos_thr=$thr --neg_thr=$thr 
--sigma_thr=$sig_thr --cutoff_hz=$cutoff --refractory_period=$refr 
--leak_rate_hz=0 --shot_noise_rate_hz=0  --vid_orig=inten --dvs_vid=dvs 
--unique_output --dvs_aedat2=dvs --output_width=346 --output_height=260 
--batch=1 --disable_slomo --dt=$dt --contrast=$contrast --dvs_exposure 
$exp --shot_noise_rate_hz=1 --show_dvs_model_state all 
--save_dvs_model_state --synthetic_input=scripts.gradients
#v2e  --output_folder output/$seq-csdvs --cs_lambda_pixels=$lambda 
--cs_tau_p_ms=$tau --pos_thr=$thr --neg_thr=$thr --sigma_thr=$sig_thr 
--cutoff_hz=$cutoff --refractory_period=$refr --leak_rate_hz=0 
--shot_noise_rate_hz=0  --vid_orig=inten --dvs_vid=dvs --unique_output 
--dvs_aedat2=dvs --output_width=346 --output_height=260 --batch=1 
--disable_slomo --dt=$dt --contrast=$contrast --dvs_exposure $exp 
--shot_noise_rate_hz=1 --show_dvs_model_state all --save_dvs_model_state 
--synthetic_input=scripts.gradients

# cloudy sky
seq=cloudy-sky
input=input/CloudySky_Slomo.mp4
lambda=10
tau=2
slomo=33

#v2e --output_folder=output/$seq-dvs --input=$input 
--input_slowmotion_factor=$slomo --disable_slomo  --pos_thr=$thr 
--neg_thr=$thr --sigma_thr=$sig_thr --cutoff_hz=$cutoff 
--refractory_period=$refr --leak_rate_hz=0 --vid_orig=inten 
--dvs_vid=dvs --unique_output --dvs_aedat2=dvs --dvs346 
--dvs_exposure=source
#v2e --cs_lambda_pixels=$lambda --cs_tau_p_ms=$tau 
--output_folder=output/$seq-csdvs --input=$input 
--input_slowmotion_factor=$slomo --disable_slomo  --pos_thr=$thr 
--neg_thr=$thr --sigma_thr=$sig_thr --cutoff_hz=$cutoff 
--refractory_period=$refr --cutoff_hz=100 --leak_rate_hz=0 
--vid_orig=inten --dvs_vid=csdvs --unique_output --dvs_aedat2=csdvs 
--dvs346  --dvs_exposure=source

tobidelbruck, 2022-08-03:
> [ { ***@***.***": "http://schema.org", ***@***.***": "EmailMessage",
> "name": "View Issue" }, "description": "View this Issue on GitHub",

Thanks to your debugging I now updated the arguments for v2e on the 
master branch to correctly show the camera size options and renamed the 
--davis_output to --ddd_output.

The colab scripts should be updated to change davis_output to ddd_output 
next otherwise we will get errors.

Note that all output size options are now in one argparse help group 
now;  hopefully this change will make things more clear.

I will try to push a change to the colab scripts if I can figure out how 
to update them. Maybe Yuhuang can help.

Tobi

gwgknudayanga, 2022-08-04:
Dear Tobi,

Thank you so much !