SensorsINI/v2e #44: moving_dot with --synthetic_input argument did not generated any video

gwgknudayanga, 2022-09-13:
Hi, 
I have executed the following command to generate synthesized video for moving dot. But in the result folder, it has video files which are zero in size. Is there anything that i need to modify? Thank you .
v2e --leak_rate=0 --shot=0 --cutoff_hz=300 --sigma_thr=.08 --pos_thr=.15 --neg_thr=.15 --dvs_exposure duration .01 --output_folder particles-slightly-less-faint-fast-2-particles --unique_output --dvs_aedat2=particles --output_width=346 --output_height=260 --batch=64 --disable_slomo --synthetic_input=scripts.moving_dot --t_total=3 --contrast=1.15 --radius=.3 --dt=0.001 --num_particles=2  --ignore-gooey

Thanks and Rgds,
Udayanga

tobidelbruck, 2022-09-13:
I haven't checked that script for some time but please post here (maybe 
as file attachment) the whole output. If I can find the time I will try 
to run it too on my home PC that is all setup for v2e.

What release/branch/commit are you running exactly?

Tobi

gwgknudayanga, 2022-09-19:
i checked out it on last Monday (12th September) actually.

tobidelbruck, 2022-09-19:
Ok, I will see if I can tell what is wrong with moving_dot in current 
master ASAP. In the meantime you are welcome to post logging output from 
your run.

tobidelbruck, 2023-07-26:
I have used synthetic input recently without problems, so closing this issue

closed 2023-07-26 by tobidelbruck