SensorsINI/v2e #45: Setting arbitrary output size for v2e output dimension

gwgknudayanga, 2022-09-19:
Hi,
Is it possible to set the output size of the v2e as 100x100 instead of the dimension of --dvs346?
Actually i need to generate events for events of size 100x100 dimension. 
Thanks !

tobidelbruck, 2022-09-19:
Absolutely, you can use --output_width and --output_height. However note 
that you also need to adapt the aedaat output class to write an 
AEDAT-2.0 format, or else just use a subspace of the DVS128 class.

Tobi

gwgknudayanga, 2022-09-19:
Thanks Tobi,
I have added it like this to the file.
SUPPORTED_SIZES=((346,260),(240,180),(100,100))
And then run again the command. But still it did not run correctly. Do i need to build the code or something like that?
If it is how should i proceed? . Thank you !
Thanks and Rgds,
Udayanga

tobidelbruck, 2022-09-19:
Just below SUPPORTED_SIZES in the file you can see the switch statement. 
Please try to look at code before posting.

gwgknudayanga, 2022-09-20:
> 

Sorry for that , Thank you !

tobidelbruck, 2022-10-11:
What output do you want? Please be more specific and provide logging output.

If you want AEDAT-2.0 format you will need to modify /add to the 
aedat_output.py. For text output you shouldn't need to change anything. 
Just disable the aedat output (please read the --help output)

gwgknudayanga, 2022-10-12:
Thanks Tobi. I disabled the aedat output and used h5 output.

tobidelbruck, 2023-07-26:
It is very easy to clone the DVS640 class in jAER to use jAER to play with the output of v2e. You just need to make the AER address format compatible.  It is pure DVS output so very simple to pack into 32-bit int address field supported by AEDAT-2.0

closed 2023-07-26 by tobidelbruck