SensorsINI/v2e #48: Incorrect x,y order in output text file

EMJian, 2022-09-30:
Hello,tobi!
x, y order is incorrect for output text file.

I run the following
`v2e.py -i data/input/tennis.mov --overwrite --timestamp_resolution=.003 --auto_timestamp_resolution=False --dvs_exposure duration 0.005 --output_folder=data/output/tennis2 --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2=None --dvs_h5=dvs.h5 --dvs_text=dvs.txt --output_width=346 --output_height=260`
Now width and height of my input are [346, 260]
But in output 'dvs.txt', x.max() is 259 and y.max() is 345.

duguyue100, 2022-09-30:
This is actually correct, the index should start from 0 and end with `height-1` or `width-1`.

EMJian, 2022-10-01:
Maybe my earlier statement was confusing.
Here's what I really want to say：
In text file, events is oder in t, x, y, p.
But it outputs text file in order t, y, x, p incorrectly.
Also i agree with your opinion of coordinate index.

tobidelbruck, 2022-10-02:
Arghh, this is annoying, sorry about this.  In looking through our own code we did a crappy job commenting which index is which for x and y. There could perhaps be a bug although in [ae_text_output.py](https://github.com/SensorsINI/v2e/blob/master/v2ecore/output/ae_text_output.py) line 75 it reads as follows which makes the order t,x,y,p
`              self.file.write('{} {} {} {}\n'.format(t[i],x[i],y[i],p[i])) # todo there must be vector way`
x is defined by line 69 as element 2 of events:
`          x = events[:, 2].astype(np.int32) # Issue #37, thanks Mohsi Jawaid`
and everything in [emulator.py](https://github.com/SensorsINI/v2e/blob/master/v2ecore/emulator.py) is based on the incoming video frame, which in line 559 has the comment
```
       new_frame: np.ndarray
            [height, width], NOTE y is first dimension, like in matlab the column, x is 2nd dimension, i.e. row.
```
which implies that element 2 *is* x and element 1 is y. Therefore I cannot see where the bug you report comes from. I must be missing something.

Did you check that you properly defined --output_width=346 and --output_height=260?  Or else used --dvs346 option?

tobidelbruck, 2022-10-02:
It could well be there is a bug here. I never use the text output myself, only AEDAT-2.0 for jAER. So let me try to find time to reproduce the problem. In the meantime you can just try swapping x and y at line 75 of ae_text_output.py

EMJian, 2022-10-04:
I checked all options. There is no doubt that all of these options are properly defined. Actually, I added the additional option "--dvs_text=dvs.txt" to the command on line 409 in [README.md](https://github.com/SensorsINI/v2e/edit/master/README.md). Of course, the rest is unchanged. This problem was still reproduced.
```bash
python v2e.py -i input/tennis.mov --overwrite --timestamp_resolution=.003 --auto_timestamp_resolution=False --dvs_exposure duration 0.005 --output_folder=output/tennis --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs_aedat2 tennis.aedat --output_width=346 --output_height=260 --stop_time=3 --cutoff_hz=15 --dvs_text=dvs.txt
```

EMJian, 2022-10-04:
For me, I don't change the v2e code. I try to swap x and y order in text file to generate the correct results.

tobidelbruck, 2022-10-05:
The earlier issue #37 was incorrectly addressed by changing the order in ae_text_output.py. The #37 patch was the result of incorrect docstring return spec in emulator.py that claimed the return order was [t,y,x,p] - this is incorrect. 
I have fixed all docstring I could find and added comments in code about events structure column ordering of t,x,y,p now.

closed 2022-10-05 by tobidelbruck