SensorsINI/v2e #50: !$final_v2e_command gives error

yazidoudou18, 2023-02-27:
v2e -i /content/v2e_tutorial_video.avi -o /content/v2e-output --overwrite --unique_output_folder false --dvs_h5 events.h5 --dvs_aedat2 None --dvs_text None --no_preview --dvs_exposure duration .033 --input_frame_rate 30 --input_slowmotion_factor 1 --disable_slomo --auto_timestamp_resolution false --pos_thres 0.2 --neg_thres 0.2 --sigma_thres 0.03 --cutoff_hz 30 --leak_rate_hz 0.1 --shot_noise_rate_hz 5 --dvs346

ERROR:v2ecore.emulator:Output file exception "Unable to create file (unable to lock file, errno = 11, error message = 'Resource temporarily unavailable')" (maybe you need to specify a supported DVS camera type?)
Traceback (most recent call last):
  File "/usr/local/bin/v2e", line 8, in <module>
    sys.exit(main())
  File "/usr/local/bin/v2e.py", line 530, in main
    emulator = EventEmulator(
  File "/usr/local/lib/python3.8/dist-packages/v2ecore/emulator.py", line 301, in __init__
    raise e
  File "/usr/local/lib/python3.8/dist-packages/v2ecore/emulator.py", line 274, in __init__
    self.dvs_h5 = h5py.File(path, "w")
  File "/usr/local/lib/python3.8/dist-packages/h5py/_hl/files.py", line 424, in __init__
    fid = make_fid(name, mode, userblock_size,
  File "/usr/local/lib/python3.8/dist-packages/h5py/_hl/files.py", line 196, in make_fid
    fid = h5f.create(name, h5f.ACC_TRUNC, fapl=fapl, fcpl=fcpl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 116, in h5py.h5f.create
OSError: Unable to create file (unable to lock file, errno = 11, error message = 'Resource temporarily unavailable')

tobidelbruck, 2023-02-27:
I haven't seen that error but do not ever use h5 output. Does it work 
with aedat output format?

duguyue100, 2023-02-28:
@yazidoudou18 feels like the HDF5 is trying to create a file but it couldn't.

Can you try a dummy example something like

```python
import h5py

file = h5py.File("my_new_file", "w")
```

and see if this triggers any errors. Also, the path of the result folder, does it allow you to write files?

yazidoudou18, 2023-02-28:
@duguyue100 The example was able to produce a file without triggering any errors. However, the 'final_v2e_command' was not able to output any files and caused the same errors.

tobidelbruck, 2023-03-01:
I can try to reproduce it this weekend, but I don't have v2e env setup 
right now. We are fighting paper deadline here.

Does the aedat and text output work correctly for you? Just turn off h5 
output to check and turn on aedat and text outputs. Also you should 
probably not specify --no_preview (or whatever disables video preview) 
to check the output visually.

yazidoudou18, 2023-03-01:
Thank you, Tobi. I have not yet set up the v2e environment. All the errors I have encountered so far are based on using v2e in Google Colab.

tobidelbruck, 2023-03-02:
You're welcome. I didn't have a chance to try anything with v2e yet. I 
hope to find some time this weekend. I think putting a breakpoint there 
would probably show the problem in 5 minutes, but I think colab is not 
good for this kind of debugging? I bet it is just some restriction in 
colab about creating files?

Does aedat2 and text output work in colab? The paths are probably 
derived by the same chunk of code.

tobidelbruck, 2023-07-26:
I just tried the below command and it run to completion with the significant warning that the IIR lowpass filter is definitely not accurate for default cutoff of 300 Hz

````
v2e -i input/tennis.mov -o output/tennis --overwrite --unique_output_folder false --dvs_h5 events.h5 --dvs_aedat2 None --dvs_text None --no_preview --dvs_exposure duration .033 --input_frame_rate 30 --input_slowmotion_factor 1 --disable_slomo --auto_timestamp_resolution false --pos_thres 0.2 --neg_thres 0.2 --sigma_thres 0.03 --cutoff_hz 30 --leak_rate_hz 0.1 --shot_noise_rate_hz 5 --dvs346
````

tobidelbruck, 2023-07-26:
Note that this was in the cloned local v2e env, not the colab workspace.

closed 2023-07-26 by tobidelbruck