SensorsINI/v2e #51: Collab requirements fail to install

neworderofjamie, 2023-05-04:
Very odd but pip install seems to decide to try and install a descending sequence of OpenCV versions:
```
Cloning into 'v2e'...
remote: Enumerating objects: 3225, done.
remote: Counting objects: 100% (995/995), done.
remote: Compressing objects: 100% (143/143), done.
remote: Total 3225 (delta 916), reused 905 (delta 852), pack-reused 2230
Receiving objects: 100% (3225/3225), 34.33 MiB | 39.95 MiB/s, done.
Resolving deltas: 100% (2381/2381), done.
/content/v2e
Looking in indexes: https://pypi.org/simple, https://us-python.pkg.dev/colab-wheels/public/simple/
Processing /content/v2e
  Preparing metadata (setup.py) ... done
Collecting numpy==1.20
  Downloading numpy-1.20.0.zip (8.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 8.0/8.0 MB 63.0 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: argcomplete in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (3.0.8)
Requirement already satisfied: engineering-notation in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (0.8.0)
Requirement already satisfied: tqdm in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (4.65.0)
Requirement already satisfied: opencv-python in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (4.7.0.72)
Requirement already satisfied: h5py in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (3.8.0)
Requirement already satisfied: torch in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (2.0.0+cu118)
Requirement already satisfied: torchvision in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (0.15.1+cu118)
Requirement already satisfied: numba in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (0.56.4)
Requirement already satisfied: matplotlib in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (3.7.1)
Collecting plyer
  Downloading plyer-2.1.0-py2.py3-none-any.whl (142 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 142.3/142.3 kB 18.1 MB/s eta 0:00:00
Collecting screeninfo
  Downloading screeninfo-0.8.1-py3-none-any.whl (12 kB)
Collecting easygui
  Downloading easygui-0.98.3-py2.py3-none-any.whl (92 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 92.7/92.7 kB 13.0 MB/s eta 0:00:00
Requirement already satisfied: scikit-image in /usr/local/lib/python3.10/dist-packages (from v2e==1.5.1) (0.19.3)
Requirement already satisfied: pyparsing>=2.3.1 in /usr/local/lib/python3.10/dist-packages (from matplotlib->v2e==1.5.1) (3.0.9)
Requirement already satisfied: kiwisolver>=1.0.1 in /usr/local/lib/python3.10/dist-packages (from matplotlib->v2e==1.5.1) (1.4.4)
Requirement already satisfied: contourpy>=1.0.1 in /usr/local/lib/python3.10/dist-packages (from matplotlib->v2e==1.5.1) (1.0.7)
Requirement already satisfied: pillow>=6.2.0 in /usr/local/lib/python3.10/dist-packages (from matplotlib->v2e==1.5.1) (8.4.0)
Requirement already satisfied: fonttools>=4.22.0 in /usr/local/lib/python3.10/dist-packages (from matplotlib->v2e==1.5.1) (4.39.3)
Requirement already satisfied: python-dateutil>=2.7 in /usr/local/lib/python3.10/dist-packages (from matplotlib->v2e==1.5.1) (2.8.2)
Requirement already satisfied: packaging>=20.0 in /usr/local/lib/python3.10/dist-packages (from matplotlib->v2e==1.5.1) (23.1)
Requirement already satisfied: cycler>=0.10 in /usr/local/lib/python3.10/dist-packages (from matplotlib->v2e==1.5.1) (0.11.0)
Requirement already satisfied: setuptools in /usr/local/lib/python3.10/dist-packages (from numba->v2e==1.5.1) (67.7.2)
Requirement already satisfied: llvmlite<0.40,>=0.39.0dev0 in /usr/local/lib/python3.10/dist-packages (from numba->v2e==1.5.1) (0.39.1)
Collecting opencv-python
  Downloading opencv_python-4.7.0.68-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (61.8 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.8/61.8 MB 13.4 MB/s eta 0:00:00
  Downloading opencv_python-4.6.0.66-cp36-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (60.9 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.9/60.9 MB 12.2 MB/s eta 0:00:00
  Downloading opencv_python-4.5.5.64-cp36-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (60.5 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.5/60.5 MB 11.3 MB/s eta 0:00:00
  Downloading opencv_python-4.5.5.62-cp36-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (60.4 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.4/60.4 MB 12.9 MB/s eta 0:00:00
  Downloading opencv_python-4.5.4.60-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (60.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.3/60.3 MB 12.5 MB/s eta 0:00:00
  Downloading opencv_python-4.5.4.58-cp310-cp310-manylinux2014_x86_64.whl (60.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 60.3/60.3 MB 13.2 MB/s eta 0:00:00
  Downloading opencv-python-4.5.3.56.tar.gz (89.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 89.2/89.2 MB 11.8 MB/s eta 0:00:00
  error: subprocess-exited-with-error
  
  × pip subprocess to install build dependencies did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Installing build dependencies ... error
error: subprocess-exited-with-error

× pip subprocess to install build dependencies did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
/content
/content
```

duguyue100, 2023-05-04:
This did not even make it through the OpenCV installation. I have a feeling colab has some problem with it

duguyue100, 2023-05-04:
@neworderofjamie I see Colab has updated their environment to 3.10, which might be the reason?

I could debug a bit later, probably faster if you can mess around.

sanjayaherath, 2023-05-07:
Got the same issue. Any fixes?

duguyue100, 2023-05-10:
I don't have the capacity yet for fixing this problem. seems like a python version thing.

mariusschaab, 2023-05-15:
Opencv-python has the minimal NumPy requirement of Version 1.21.2 for Python 3.10 [see here](https://github.com/opencv/opencv-python/blob/6b73d90fc3e50ba6858926d299b49f0228e19d68/setup.py#L31). That's conflicting with the Version 1.20 specified in [setup.py](https://github.com/SensorsINI/v2e/blob/7cb90cb828e80b25114bc3dc60d4ae347f6fef3b/setup.py#L37). 

You can create a copy of this CoLab and change the NumPy version to 1.22 after the `git clone` step in `/content/v2e/setup.py` line 37. Then run `pip install .` again.

neworderofjamie, 2023-05-18:
Thanks @tobidelbruck - it now installs but I get an error when it tries to run the command:
```
ERROR:v2e:There is no synthetic input class specified but there are extra arguments ['--davis_output'] that are probably incorrect
```

duguyue100, 2023-05-18:
@neworderofjamie I can confirm this, let me check what's the error there. probably a misplaced if statement.

tobidelbruck, 2023-05-18:
Arghh, sorry about the hassles. Yuhuang if you don't have time, let me 
know. I can try this evening I hope.

duguyue100, 2023-05-18:
@neworderofjamie You can delete and restart the colab runtime and give it a try, it should work now.

@tobidelbruck The troubled code is here:
https://github.com/SensorsINI/v2e/blob/master/v2e.py#L199-L201

```
     if synthetic_input_instance is None and len(other_args)>0:
         logger.error(f'There is no synthetic input class specified but there are extra arguments {other_args} that are probably incorrect')
         v2e_quit(1)
```

It looks like you introduced this 7 months ago. I'm not sure what's the reason (probably something to do with synthetic input)
I commented it out, and at least the colab script works.

neworderofjamie, 2023-05-18:
Awesome! it now works - thanks @duguyue100

closed 2023-05-18 by duguyue100

tobidelbruck, 2023-05-18:
No, I think the colab example has some incorrect (or modified) arguments 
that argparse does not recognize and are passed to the other_args for a 
synthetic image generator to (maybe) use. If you print "other_args" or 
look at the log output it will say what those are.

But it works now. However one or more arguments in the colab are 
probably ignored.