SensorsINI/v2e #52: it seems like the code do not support --dvs640.

menyunlei, 2023-05-07:
it seems like the code do not support --dvs640   whenever i insert--dvs640, it will out put ValueError: AEDAT-2.0 output width=640 height=480 not supported; add your camera to v2ecore.output.aedat2_output or use one of the predefined DVS cameras, e.g. --dvs346 or --dvs240 that have sizes ((346, 260), (240, 180))

NPU-YanChi, 2023-07-25:
I meet the same question

NPU-YanChi, 2023-07-25:
Thanks for your quick reply.
I temporarily solved this problem by commenting on the parameters associated with aedat2.

tobidelbruck, 2023-07-25:
I added the --dvs640 option to aedat2_output.py, and tested it with jAER 
DVS640 AEChip class. It seems to work. I also fixed a bug with 
nonmonotonic timestamps using original shot noise events where the 
random permutation would cause the timestamps to go out of order. Please 
pull changes to get these fixes.

closed 2023-07-26 by tobidelbruck