SensorsINI/v2e #56: No module named 'dv_processing'

Ikhadija-5, 2023-08-14:
Hello, I am trying to run the Colab version of v2e. However when I get to the cell with the command `!$final_v2e_command`, I encounter the error below: Can you please help me fix this?

Traceback (most recent call last):
  File "/usr/local/bin/v2e", line 5, in <module>
    from v2e import main
  File "/usr/local/bin/v2e.py", line 34, in <module>
    from v2ecore.v2e_args import v2e_args, write_args_info, SmartFormatter
  File "/usr/local/lib/python3.10/dist-packages/v2ecore/v2e_args.py", line 6, in <module>
    from v2ecore.emulator import EventEmulator
  File "/usr/local/lib/python3.10/dist-packages/v2ecore/emulator.py", line 27, in <module>
    from v2ecore.output.aedat4_output import AEDat4Output
  File "/usr/local/lib/python3.10/dist-packages/v2ecore/output/aedat4_output.py", line 10, in <module>
    import dv_processing as dv
ModuleNotFoundError: No module named 'dv_processing'

tobidelbruck, 2023-08-14:
Argghh, I guess that is a byproduct of the new pull from Min Liu that 
includes AEDAT-4.0 support. I think if you include a "pip install 
dv-processing" it should work.

Yuhuang, do you agree?

Tobi

duguyue100, 2023-08-14:
Yes, this should work. I added this line in the Google Colab, should work now: https://colab.research.google.com/drive/1czx-GJnx-UkhFVBbfoACLVZs8cYlcr_M?usp=sharing

duguyue100, 2023-08-14:
@Ikhadija-5 Let me know if re-runing the notebook works, then I will close the issue.

Ikhadija-5, 2023-08-14:
Hi, thanks for the quick response @tobidelbruck  and @duguyue100 . It works with the pip install now. I was a bit lost as the video_path was not specified in the code, but I added it and I am able to generate the events now. Thank you.

closed 2023-08-14 by duguyue100

keyclick, 2025-01-08:
Hello there,
I know this is quite an old thread but I am able to install dv_processing on windows powershell with administrator privilages but when I try to do it with jupyter or conda environment, I receive an error.
Anybody has any idea?

```
PS D:\jupyter> pip install dv-processing 
Collecting dv-processing
  Using cached dv_processing-1.7.9.tar.gz (9.5 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [26 lines of output]
      fatal: not a git repository (or any of the parent directories): .git
      Using git hash
      Traceback (most recent call last):
        File "C:\Anaconda3\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Anaconda3\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Anaconda3\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
                 ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Ajith's PC\AppData\Local\Temp\pip-build-env-ucnj5qd8\overlay\Lib\site-packages\setuptools\build_meta.py", line 334, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\Ajith's PC\AppData\Local\Temp\pip-build-env-ucnj5qd8\overlay\Lib\site-packages\setuptools\build_meta.py", line 304, in _get_build_requires
          self.run_setup()
        File "C:\Users\Ajith's PC\AppData\Local\Temp\pip-build-env-ucnj5qd8\overlay\Lib\site-packages\setuptools\build_meta.py", line 320, in run_setup
          exec(code, locals())
        File "<string>", line 208, in <module>
        File "<string>", line 199, in get_version
        File "C:\Anaconda3\Lib\subprocess.py", line 466, in check_output
          return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Anaconda3\Lib\subprocess.py", line 571, in run
          raise CalledProcessError(retcode, process.args,
      subprocess.CalledProcessError: Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

hoangvanphi2004, 2025-02-23:
> Hello there, I know this is quite an old thread but I am able to install dv_processing on windows powershell with administrator privilages but when I try to do it with jupyter or conda environment, [...]

Hey have you solved the problem ? im having the same problem !!!

tobidelbruck, 2025-02-23:
You should post on davis-users list 
https://groups.google.com/g/davis-users, since this is a question about 
DV installation.

v2e uses DV I think only for making aedat4 output files.

keyclick, 2025-05-20:
> Hey have you solved the problem ? im having the same problem !!!

Yes. I was able to solve it.

keyclick, 2025-05-20:
> You should post on davis-users list
> https://groups.google.com/g/davis-users, since this is a question about
> DV installation.
> 
> v2e uses DV I think only for making aedat4 output files.

Thankyou. This will be great. I was looking to reduce the noise and this is like a whole library of info.