SensorsINI/v2e #58: Blank (gray) event frames in the output video

mohsij, 2023-10-23:
Hi,

I've been using the latest version of v2e and it seems that some frames are either blank (gray) or the event pixels are all less intense than the last frame. The object in the input video is moving consistently. The last good commit for me is 1edd9e3d2b30435edf4759ffc1c3703c572c73c0 where the event frames look consistent and I get no blank frames.

My run command is:
v2e -i frames --overwrite --input_frame_rate=10 \
--timestamp_resolution=0.1 --disable_slomo --auto_timestamp_resolution=False \
--dvs_exposure duration 0.1 --output_folder=event_data --overwrite --pos_thres=.15 --neg_thres=.15 \
--sigma_thres=0.3 --dvs_text events.csv --output_width=1280 --output_height=720 --cutoff_hz=30 --avi_frame_rate=10 \
--refractory_period 0 --dvs_aedat2 None

What was changed after that commit that I could get blank frames?

Thanks

mohsij, 2023-10-24:
The only problem with commit 1edd9e3d2b30435edf4759ffc1c3703c572c73c0 is that I get 717 event frames output from an input of 720 frames. With the commits after that I do get 718 but then run into the issue outlined i.e some blank frames and some with less bright pixels even though the movement is consistent.

It looks like  the renderer outputs batches of 7 frames at a time and the first frame in each batch can either be blank or with 'muted' event pixel intensities. See attached side-by-side below of frame 00006.png and frame 00007.png.
[image]

tobidelbruck, 2023-11-07:
Thanks for providing commit number. That commit only pushed changes to some synthetic input examples.

Did you check that you are using correct switch:
```
--dvs_exposure DVS_EXPOSURE [DVS_EXPOSURE ...]
                        Mode to finish DVS frame event integration:
                        	duration time: Use fixed accumulation time in seconds, e.g. 
                        		--dvs_exposure duration .005; 
                        	count n: Count n events per frame,e.g.
                        		--dvs_exposure count 5000;
                        	area_count M N: frame ends when any area of N x N pixels fills with M events, e.g.
                        		-dvs_exposure area_count 500 64
                        	source: each DVS frame is from one source frame (slomo or original, depending on if slomo is used)
```
In your case did you use `--dvs_exposure` source option?

Also, if you really disabled slomo upsampling, then 
```
 --batch_size BATCH_SIZE
                        Batch size in frames for SuperSloMo. Batch size 8-16
                        is recommended if your GPU has sufficient memory.
```
should not have any effect.