SensorsINI/v2e #60: APS and DVS time in DDD dataset are not aligned

denyunhero, 2023-10-30:
In the DDD17 and DDD20 datasets, I used v2e.dataset_scripts.ddd.ddd_extract_data.py to extract .avi and .aedat2 files from the original .hdf5 file. However, when I checked both the .avi (APS) and .aedat2(DVS) files, I found the time duration for APS and DVS is not different. For example, the .avi (APS) file for rec1487333001 lasts for 2min16sec, but that of the .aedat2(DVS) is  1min33sec.

When I checked the timestamps for the .avi(APS) file in the corresponding hdf5 file, I found the .avi(APS) file was captured in the Variable frame rates. However, the .avi file can only display with a fixed frame rate. It means that the time (1min33secs) in the .avi (file) is not accurate.

In the v2e, the events are simulated only with a fixed APS frame rate. I have not found yet that there are inputs for APS timestamps corresponding to each APS frame. 

Therefore, it seems that DDD17 and DDD20 datatsets cannot be used for event simulation in v2e. Is there something wrong with my opinion? I will be grateful for your reply.

tobidelbruck, 2023-10-30:
It's been a long time. Maybe Yuhaung can comment?  We definitely used 
DDD17/20 for studying realism of v2e emulation by using the (bad) APS 
frames to generate events and compare the event count statistics to 
adjust the DVS thresholds.

https://sites.google.com/view/video2events/home#h.p_VXJ0LTrAvAhN

https://github.com/SensorsINI/v2e/tree/master#model-parameters

https://github.com/SensorsINI/v2e/blob/master/dataset_scripts/ddd/ddd_find_thresholds.py

https://github.com/SensorsINI/v2e/blob/master/media/find_thresholds.png

Tobi

denyunhero, 2023-11-01:
> It's been a long time. Maybe Yuhaung can comment?  We definitely used DDD17/20 for studying realism of v2e emulation by using the (bad) APS frames to generate events and compare the event count [...]

Hi, Prof. Delbruck, Thanks for your kindly responding.