SensorsINI/v2e #62: Problem about downloading the Super-SloMo model.

SYuan-X, 2023-11-09:
Hello author, this is a very meaningful work. I met a small problem when runnig this code.
 When I try to download the Super-SloMo pre-trained model, 404 error happened. Can you provide another way to download the model? thank you so much!

The output on colab looks like this:
--2023-11-09 09:57:46--  https://docs.google.com/uc?export=download&confirm=&id=17QSN207h05S_b2ndXjLrqPbBTnYIl0Vb
Resolving docs.google.com (docs.google.com)... 108.177.98.139, 108.177.98.113, 108.177.98.100, ...
Connecting to docs.google.com (docs.google.com)|108.177.98.139|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-11-09 09:57:46 ERROR 404: Not Found.

LI474525323, 2023-11-09:
Yes, I can't download the file either

duguyue100, 2023-11-09:
Guys, I have the file now, will reupload and update the link soon.

duguyue100, 2023-11-09:
@SYuan-X  @LI474525323  I've updated the link in the readme and in the colab notebook. Please check it out, I will close this issue for now.

closed 2023-11-09 by duguyue100

LI474525323, 2023-11-14:
thank you very much!

 
Closed #62 as completed.