SensorsINI/v2e #69: I put different pictures together into a video and then ran v2e. There will be flickering during the transition. Is there any way to remove it?

108360215, 2024-07-18:


108360215, 2024-07-18:
[image]

duguyue100, 2024-07-18:
v2e doesn't do any video smoothing given a video. Maybe do some post-processing before you produce the video? Use some kind of smoothing or stabilization from a video editor?

tobidelbruck, 2024-07-18:
Could you be more clear about your setup?  Are the pictures from a video 
sequence? If so, is there aliasing in the video, i.e. is the frame rate 
too low to capture the highest frequencies?  If the spatial jumps are 
too large, then superslomo cannot compute optical flow.  Please share 
part of the video sequence.

duguyue100, 2024-07-19:
Do you know the original frame rate of these video frames? If it's a super low rate, the displacement between frames could be very large. In this case, you might want to remove the `--disable-slomo` option from the command to allow the SloMo model to interpolate some frames in between, otherwise, you will always get this kind of effect as the changes between pixels are very big.

108360215, 2024-07-19:
original frame rate depend what I set up, I tried to set up 30 before but still noise, I can try to use this --disable-slomo, but what do you think my output fps should be? also 30? And you say I can interpolate some frames in between, you mean when I combine the non continuous picture to video?

tobidelbruck, 2024-07-19:
Based on a quick  look at your video sequence your bursty output is 
expected and v2e works "correctly". v2e is intended to realistically 
simulate DVS events resulting from continuous time video input. If you 
feed stroboscopic image sequences, the output will in your case is 
realistic: you are showing the "camera" a series of images which have 
big changes between them, as a result there will be huge bursts of 
events.   Or am I missing something?

108360215, 2024-07-20:
So I'm curious if there's a way to merge these different types of images into a video and then convert it to event data. Otherwise, I can only find continuous images to convert, which significantly limits the size of my dataset.
What do you guys think about