SensorsINI/v2e #71: Spiking Pattern in event generation of a Focal Stack

FabianRudolph, 2024-09-01:
I have used V2E to simulate events from a focal stack. I am running it at different frequencies. However, I have the issue that there is  a repeating pattern. In theory, it should be fine using a constant duration since the field of view is changing as a constant as well. For example, for 100 frames I use an exposure duration of 0.01 which in theory should sample it at 1 "image" per event timestamp, right? I also tried to use it double the frequency (i.e. 0.02).

Now, my issue is that this results in a repeating pattern where 1 frame has a lot of events, then less and then there is an empty frame. I have provided a plot that shows the events per frame that shows it for the different frequencies where I use 50, 100, 1000 frames etc.

In my understanding, the events should be occuring all at once, as the new image is read, resulting in 100 different timestamps for a 100 image focal stack example. How can I achieve this?

100 Hz:
[image]

500 Hz:
[image]

Perhaps, do I have a misunderstanding of the exposure duration? Because when looking at the timestamps, they don't have a "constant" timestamp and are rather binned, to what seems to align with the issue.

5Hz:
[image]

500Hz:
[image]

Here is a selection of the options I've used on my images (1000 Frames example)

`
--input_frame_rate 1000 
--dvs_exposure duration .002 
--disable_slomo
--pos_thres 0.2 
--neg_thres 0.2
--sigma_thres 0.02 
--cutoff_hz 0 
--leak_rate_hz 0
--shot_noise_rate_hz 0 
--dvs1024
`

FabianRudolph, 2024-09-12:
In order to fix it, I have binned the events to my desired frequency.

jingyang-huang, 2025-06-26:
> In order to fix it, I have binned the events to my desired frequency.为了修复它，我已将事件分类到我想要的频率。

Hi Fabian, I met the same problem just like you, some total empty frame occurs after a while, can you share with me how did you fix this, and here is my running shell:
```shell
python v2e.py -i data/mine/output_1000fps_lossless.mp4  --no_preview --disable_slomo --dvs_params noisy --timestamp_resolution=.003  --auto_timestamp_resolution=False --avi_frame_rate 100  --dvs_exposure duration 0.005 --output_folder=output/mine --overwrite --pos_thres=.15 --neg_thres=.15 --sigma_thres=0.03 --dvs640

```

FabianRudolph, 2025-07-14:
I ended up binning the events into their respective intervals and to not allow them to fall in between two bins. Essentially, I sort of discretized them. Please note that this is an unnatural output for the DVS but I needed it for my usecase. You could also try changing the parameters to improve the output.